bblanchon/ArduinoJson #478: Storing JSON as a string/char in flash memory

benariamir, 2017-04-04:
Hey guys,
Quick question.
I'm using AWS IoT and need publish messages in a JSON format.
To achieve that I need it stored as a char array or a String.
But, the JSON string is too large to fit into the SRAM so I can't just
String buffer;
root.printTo(buffer);

What I do need (I think, please be patient with a newbie :) ) is to store the char array in flash memory using F() or PSTR().

But how do I achieve that with ArduinoJson ?

Thanks.

bblanchon, 2017-04-04:
Hi @benariamir,

As far as I know, you can only write to Flash at program time, not at run time.
Even if it was possible it would be very slow.

However, ArduinoJson is able to write to a `Print` implementation like `EthernetClient` or `WifiClient`.

See:

* [API Reference: JsonObject::printTo()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/printto/)
* [Examples: JsonServer.ino](https://bblanchon.github.io/ArduinoJson/example/http-server/)
* [Examples: JsonHttpClient.ino](https://bblanchon.github.io/ArduinoJson/example/http-client/)

bblanchon, 2017-04-12:
@benariamir any progress?

closed 2017-04-18 by bblanchon