bblanchon/ArduinoJson #479: Delete an object by key

mcmega, 2017-04-05:
Hello!
I'm using this library.
Input data:
[  {"KA": "Value"}, ..., {"KA": "Value"}  ]
How to delete all objects {...}, in which the value of the key KA is the same.
Help me please!

bblanchon, 2017-04-08:
Hi Oleg,

You need to loop through all elements of the root array with [JsonArray's iterator](https://bblanchon.github.io/ArduinoJson/api/jsonarray/begin_end/) and keep track of each matching index.
Then you need to call [JsonArray::removeAt()](https://bblanchon.github.io/ArduinoJson/api/jsonarray/removeat/) for each of these indexes.

It would be simpler and more efficient if `JsonArray::removeAt()` would take an iterator, but unfortunately it's not the case right now.
I'll try to add this feature soon.

mcmega, 2017-04-08:
I'm learning C++ (Arduino). Please make a sample code using the library.

bblanchon, 2017-04-12:
If you use the `master` branch of ArduinoJson you can write something like that:

```c++
#include <set>
#include <string>

// Transforms
// [{"K":"10"},{"K":"12"},{"K":"15"},{"K":"12"},{"K":"15"}]
// into
// [{"K":"10"},{"K":"12"},{"K":"15"}]
void removeDuplicates(JsonArray& array) {
  // keep a history of values
  std::set<std::string> seenValues;

  // scan all elements of the array
  for (JsonArray::iterator it = array.begin(); it != array.end(); ++it) {
    // get the object in the array
    JsonObject& element = *it;

    // insert the value in the set
    bool isNew = seenValues.insert(element["KA"]).second;

    // remove it if we've seen it already
    if (!isNew) array.remove(it);
  }
}
```

mcmega, 2017-04-13:
Does not work, I cause:
```
void DevsIP() {
  String IPList = "[{\"K\":\"10\"},{\"K\":\"12\"},{\"K\":\"15\"},{\"K\":\"12\"},{\"K\":\"15\"}]";
  removeDuplicates(IPList);
}
```
Does not compile, Included the library through the manager in Arduino. Help!
**Errors:**
```
/Users/poe/Documents/SOI-Tech/WL/WL-WiFi/HTTP.ino: In function 'void DevsIP()':
HTTP:109: error: invalid initialization of reference of type 'ArduinoJson::JsonArray&' from expression of type 'String'
   removeDuplicates(IPList);
                          ^
HTTP:113: error: in passing argument 1 of 'void removeDuplicates(ArduinoJson::JsonArray&)'
 void removeDuplicates(JsonArray& array) {
      ^
/Users/poe/Documents/SOI-Tech/WL/WL-WiFi/HTTP.ino: In function 'void removeDuplicates(ArduinoJson::JsonArray&)':
HTTP:123: error: call of overloaded 'insert(ArduinoJson::JsonObjectSubscript<const char*>)' is ambiguous
     bool isNew = seenValues.insert(element["K"]).second;
                                                ^
/Users/poe/Documents/SOI-Tech/WL/WL-WiFi/HTTP.ino:123:48: note: candidates are:
In file included from /Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/set:61:0,
                 from /Users/poe/Documents/SOI-Tech/WL/WL-WiFi/WL-WiFi.ino:11:
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:460:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(const value_type&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(const value_type& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:469:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(std::set<_Key, _Compare, _Alloc>::value_type&&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(value_type&& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:529:7: note: void std::set<_Key, _Compare, _Alloc>::insert(std::initializer_list<_Tp>) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >]
       insert(initializer_list<value_type> __l)
       ^
HTTP:126: error: 'class ArduinoJson::JsonArray' has no member named 'remove'
     if (!isNew) array.remove(it);
                       ^
exit status 1
invalid initialization of reference of type 'ArduinoJson::JsonArray&' from expression of type 'String'
```

bblanchon, 2017-04-19:
Hi Oleg,

You need to parse the array before using the function:

```c++
char json[] = "[{\"K\":\"10\"},{\"K\":\"12\"},{\"K\":\"15\"},{\"K\":\"12\"},{\"K\":\"15\"}]";

DynamicJsonBuffer jb;
JsonArray& IPList = jb.parseArray(json);
removeDuplicates(IPList);
IPList.printTo(Serial);
```

mcmega, 2017-04-19:
I'm doing in Arduino. Does not work...
"json" I have the type String.
All Code:

```
#include <set>
#include <string>
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200, SERIAL_8N1);
  Serial.println("");
}

String json = "[{\"K\":\"1\"},{\"K\":\"2\"},{\"K\":\"1\"},{\"K\":\"1\"},{\"K\":\"2\"}]";

void removeDuplicates(JsonArray& array) {
  std::set<std::string> seenValues;
  for (JsonArray::iterator it = array.begin(); it != array.end(); ++it) {
    JsonObject& element = *it;
    bool isNew = seenValues.insert(element["K"]).second;
    if (!isNew) array.remove(it);
  }
}

void loop() {
  DynamicJsonBuffer jb;
  JsonArray& IPList = jb.parseArray(json);
  removeDuplicates(IPList);
  
//  Serial.println(IPList);
}
```
Errors:

```
Arduino: 1.8.1 (Mac OS X), Плата:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 921600, 1M (512K SPIFFS), nodemcu, Disabled, None"

/Users/poe/Documents/Arduino/dubl/dubl.ino: In function 'void removeDuplicates(ArduinoJson::JsonArray&)':
dubl:16: error: call of overloaded 'insert(ArduinoJson::JsonObjectSubscript<const char*>)' is ambiguous
     bool isNew = seenValues.insert(element["K"]).second;
                                                ^
/Users/poe/Documents/Arduino/dubl/dubl.ino:16:48: note: candidates are:
In file included from /Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/set:61:0,
                 from /Users/poe/Documents/Arduino/dubl/dubl.ino:1:
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:460:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(const value_type&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(const value_type& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:469:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(std::set<_Key, _Compare, _Alloc>::value_type&&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(value_type&& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:529:7: note: void std::set<_Key, _Compare, _Alloc>::insert(std::initializer_list<_Tp>) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >]
       insert(initializer_list<value_type> __l)
       ^
dubl:17: error: 'class ArduinoJson::JsonArray' has no member named 'remove'
     if (!isNew) array.remove(it);
                       ^
exit status 1
call of overloaded 'insert(ArduinoJson::JsonObjectSubscript<const char*>)' is ambiguous
```

mcmega, 2017-04-19:
I tried in xCode
```
#include <iostream>
#include <typeinfo>
#include <set>
#include <string>
#include "ArduinoJson.h"

void removeDuplicates(JsonArray& array) {
    std::set<std::string> seenValues;
    for (JsonArray::iterator it = array.begin(); it != array.end(); ++it) {
        JsonObject& element = *it;
        
        bool isNew = seenValues.insert(element["K"]).second;
        if (!isNew) array.remove(it);
        std::cout << element["K"] << " - " << typeid(element["K"]).name() << std::endl;
    }
}

int main() {
    char json[] = "[{\"K\":\"1\"},{\"K\":\"2\"},{\"K\":\"1\"},{\"K\":\"1\"},{\"K\":\"2\"}]";
    DynamicJsonBuffer jb;
    JsonArray& IPList = jb.parseArray(json);
    removeDuplicates(IPList);
    
    std::cout << IPList << std::endl;
    return 0;
}
```
Error: Call to member function 'insert' is ambiguous on line  bool isNew = ...

Type value element["K"] - N11ArduinoJson19JsonObjectSubscriptIPKcEE

mcmega, 2017-04-19:
Does not work, platform Arduino ESP8266
```
#define ARDUINOJSON_ENABLE_ARDUINO_STRING 1

#include <set>
#include <string>
#include <ArduinoJson.h>

void setup() {
  UART_Init(115200);
}

void UART_Init(int speed) {             // ИНИЦИАЛИЗАЦИЯ UART
  Serial.begin(speed, SERIAL_8N1);      // Настраиваем интерфейс UART
  Serial.println("");
}

void removeDuplicates(JsonArray& array, std::string Key) {
    std::set<std::string> seenValues;
    for (JsonArray::iterator it = array.begin(); it != array.end(); ++it) {
        JsonObject& element = *it;
        bool isNew = seenValues.insert(element[Key]).second;
        if (!isNew) array.remove(it);
    }
}

String json = "[{\"K\":\"1\"},{\"K\":\"1\"},{\"K\":\"2\"},{\"K\":\"2\"},{\"K\":\"2\"},{\"K\":\"1\"},{\"K\":\"1\"},{\"K\":\"2\"},{\"K\":\"2\"}]";
DynamicJsonBuffer jb;                     // Создаем динамический буфер jsonBuffer
JsonArray& IPList = jb.parseArray(json);  // Парсим строку json и создаем из нее массив IPList
String jsonStr = "";                      // Создаем строковую переменную для хранения конечного результата

void loop() {
  removeDuplicates(IPList,"K");           // Удаление объектов из массива IPList, если значению ключа K одинаковое
  IPList.printTo(jsonStr);                // Преобразуем json в строку
  Serial.println(jsonStr);                // Выводим в UART
}
```
Errors:

```
Arduino: 1.8.1 (Mac OS X), Плата:"Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 921600, 1M (512K SPIFFS), nodemcu, Disabled, None"

/Users/poe/Documents/Arduino/removeDuplicates/removeDuplicates.ino: In function 'void removeDuplicates(ArduinoJson::JsonArray&, std::string)':
removeDuplicates:20: error: call of overloaded 'insert(ArduinoJson::TypeTraits::EnableIf<true, ArduinoJson::JsonObjectSubscript<const std::basic_string<char>&> >::type)' is ambiguous
         bool isNew = seenValues.insert(element[Key]).second;
                                                    ^
/Users/poe/Documents/Arduino/removeDuplicates/removeDuplicates.ino:20:52: note: candidates are:
In file included from /Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/set:61:0,
                 from /Users/poe/Documents/Arduino/removeDuplicates/removeDuplicates.ino:3:
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:460:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(const value_type&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(const value_type& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:469:7: note: std::pair<typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator, bool> std::set<_Key, _Compare, _Alloc>::insert(std::set<_Key, _Compare, _Alloc>::value_type&&) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >; typename std::_Rb_tree<_Key, _Key, std::_Identity<_Key>, _Compare, typename _Alloc::rebind<_Key>::other>::const_iterator = std::_Rb_tree_const_iterator<std::basic_string<char> >; std::set<_Key, _Compare, _Alloc>::value_type = std::basic_string<char>]
       insert(value_type&& __x)
       ^
/Users/poe/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_set.h:529:7: note: void std::set<_Key, _Compare, _Alloc>::insert(std::initializer_list<_Tp>) [with _Key = std::basic_string<char>; _Compare = std::less<std::basic_string<char> >; _Alloc = std::allocator<std::basic_string<char> >]
       insert(initializer_list<value_type> __l)
       ^
removeDuplicates:21: error: 'class ArduinoJson::JsonArray' has no member named 'remove'
         if (!isNew) array.remove(it);
                           ^
exit status 1
call of overloaded 'insert(ArduinoJson::TypeTraits::EnableIf<true, ArduinoJson::JsonObjectSubscript<const std::basic_string<char>&> >::type)' is ambiguous
```

bblanchon, 2017-04-22:
You're welcome Oleg!
Congratulation for fixing the code by yourself 👍 
The library manager will be updated in a few days, once everything is ready for 5.9.0.

closed 2017-04-22 by bblanchon