bblanchon/ArduinoJson #480: Unable to parseObject after API call

thiervoj, 2017-04-11:
Hello,

I'm having a problem using `parseObject` after an HTTP call to an API.
I modified the example `JsonHttpClient` given with the package to fit with the API that I want to call and the data that I want to get.

Here is my code in the `readReponseContent` function :

```C++
DynamicJsonBuffer jsonBuffer;
  
  JsonObject& root = jsonBuffer.parseObject(client);

  if (!root.success()) {
    Serial.println("JSON parsing failed!");
    return false;
  }
```

The problem is that the code always goes in the `if` statement.

Here is the json that i'm receiving from the client:
```json
{"status":"ok","data":{"aqi":46,"idx":3056,"attributions":[{"url":"http://www.air-rhonealpes.fr/","name":"Observatoire Air Rhone-Alpes"}],"city":{"geo":[45.895344907131,6.118434383291],"name":"Annecy Loverchy","url":"http://aqicn.org/city/france/rhonealpes/haute-savoie/annecy-loverchy/"},"dominentpol":"pm25","iaqi":{"h":{"v":45},"no2":{"v":3.2},"o3":{"v":45.1},"p":{"v":1020},"pm10":{"v":20},"pm25":{"v":46},"t":{"v":18.11}},"time":{"s":"2017-04-11 17:00:00","tz":"+01:00","v":1491930000}}}
```

I also tried to use a StaticJsonBuffer but I only got some strange signs as if the content that I want to print wasn't in UTF-8, yet it is.

To debug I tried to `Serial.print` using `client.readStringUntil('\n')` and the json response is correctly received by my arduino board.

What am I doing wrong here ? Have I forgot to do something ?

bblanchon, 2017-04-12:
Hello Jordan,

Two things come to mind:

1. You may not have enough memory
    Verify that you have enough RAM and set the appropriate size for the `JsonBuffer`
2. You may have [Chunked transfer encoding](https://fr.wikipedia.org/wiki/Chunked_transfer_encoding) prefix.
    Verify that you are in HTTP 1.0 or skip the size prefix.

Have a look at:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)

Regards,
Benoit

thiervoj, 2017-04-12:
Hi Benoit,

I'm using a `DynamicJsonBuffer`, do I still have to set a size for the `jsonBuffer` ?
I tried to give it a size following [this link](https://bblanchon.github.io/ArduinoJson/assistant/) but the problem persists.

My Arduino is a Uno so it has an ATmega328 chip with 32kb of Flash memory and 2kb of SRAM.

And I'm already using HTTP 1.0...

bblanchon, 2017-04-12:
The [Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) says you need 779 bytes including the duplication of the input.
It's tight on a 2KB device but it should work.

Make sure you don't have any big variable in memory (watch for strings in particular) and then try with `StaticJsonBuffer<900>`.

thiervoj, 2017-04-12:
Ok it works now, I was not using the same "Additional bytes" value as the one given in the assistant.

Thank you for your quick answer.
And nice job with this package 👍  !

closed 2017-04-12 by thiervoj