bblanchon/ArduinoJson #481: Parsing fails

bridges23, 2017-04-16:
Hello.

I have read all necessary documentation and followed your example but I still face the problem that parsing fails for the following lines of code on my Arduino Ethernet, would be amazing if you give me a hint on where I can look for the error:

    DynamicJsonBuffer jsonBuffer(250);
    const char* json = "{\"Header\":\"Selectionnez...\",\"Products\"\"klein\",\"gross\"],\"Time\":1492374466}";

    JsonObject& root = jsonBuffer.parseObject(json);
    
root.success() always returns false.

Thanks a lot for the input.
Steffen

bblanchon, 2017-04-17:
Hello,

The input is not a valid JSON:

```json
{"Header":"Selectionnez...","Products""klein","gross"],"Time":1492374466}
```

See:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)

Regards,
Benoit

bridges23, 2017-04-18:
Hi Benoit.

Sorry there was a hickup during copy/paste. I am using the correct JSON and I even boiled it down to 
   "JsonObject& root = jsonBuffer.parseObject("{\"Header\":\"test\"}");"

in order to test whether the JSON is the problem, it is not. 

Something goes wrong here during parsing but I am not familiar with your library enough to debug it myself. Do you have another idea of a potential problem source resp. can you give ma a hint on where to debug?

Thanks a lot!!
Steffen

bblanchon, 2017-04-18:
Hi Steffen,

Does the [JsonParserExample.ino](https://bblanchon.github.io/ArduinoJson/example/parser/) works?
Does the program generated by the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) works?

bridges23, 2017-04-18:
Hi Benoit.

Both examples work - so I guess it must be a memory issue!? I am trying to get your RFID Drinks-Terminal Project running on Arduino Ethernet by the way. 
Although this really surprises me as I use the Dynamic buffer and if I check the available memory with that little piece of code 

```
int freeRam()
{
  extern int __heap_start, *__brkval;
  int v;
  return(int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval);
  }
```

the result is 300+

Do you have any other idea on how I can get this running? 

Thanks so much.
Steffen

bblanchon, 2017-04-19:
Hi Steffen,

The RFID Terminal was made with an older version of the library but should work with 5.8.
Unfortunately I don't have the hardware anymore so I cannot update the source code reliably.

Arduino Ethernet has very little RAM, so you need to be very carefull.
My advice: **don't use dynamic memory allocation.**
In particular: don't use `DynamicJsonBuffer` and `String`.
Instead use `StaticJsonBuffer` and `char[]` of the right size.

See:

* [Memory Model](https://bblanchon.github.io/ArduinoJson/doc/memory/)
* [What are the differences between StaticJsonBuffer and DynamicJsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer/)
* [Parse JSON efficiently](http://blog.benoitblanchon.fr/arduino-json-parser/)

bridges23, 2017-04-20:
Thanks Benoit. That is indeed the problem. I really wonder how you got along in that project with such little memory but I will simplify the communication between the drinks server and the arduino!  THANKS very much for the library and your help.

closed 2017-04-20 by bridges23