bblanchon/ArduinoJson #482: Parsing JSON Crash Arduino Mega

marckaraujo, 2017-04-17:
Hello, I really have tried a lot of things to solve this, but I nothing worked.

This is my json:
```
[
  {
    "codeValue": null,
    "kind": "IR",
    "bits": 32,
    "repeatValue": false,
    "codeRaw": [
      "3450",
      "1700",
      " 450",
      "450",
      " 450",
      "1250",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "450",
      " 400",
      "450",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "450",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "1250",
      " 450",
      "450",
      " 400",
      "450",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 500",
      "1250",
      " 450",
      "1250",
      " 500",
      "350",
      " 500",
      "1250",
      " 450",
      "400",
      " 450",
      "400",
      " 500",
      "350",
      " 500",
      "350",
      " 500",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 500",
      "350",
      " 500",
      "1250",
      " 450",
      "400",
      " 450",
      "1250",
      " 500",
      "350",
      " 500",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 450",
      "400",
      " 500",
      "1250",
      " 450",
      "400",
      " 450",
      "1250",
      " 500",
      "1200",
      " 500",
      "1250",
      " 450",
      "400",
      " 500",
      "1200",
      " 500"
    ],
    "codeType": -1
  },
  {
    "kind": "DELAY",
    "time": 1
  },
  {
    "codeValue": null,
    "kind": "IR",
    "bits": 32,
    "repeatValue": false,
    "codeRaw": [
      "350",
      "600",
      " 350",
      "600",
      " 300",
      "300",
      " 650",
      "600",
      " 350",
      "300",
      " 600",
      "300",
      " 650",
      "300",
      " 650",
      "600",
      " 300",
      "300",
      " 650",
      "300",
      " 650",
      "250",
      " 650",
      "600",
      " 350",
      "300",
      " 650",
      "600",
      " 300",
      "300",
      " 650",
      "600",
      " 350"
    ],
    "codeType": -1
  },
  {
    "kind": "DELAY",
    "time": 1
  },
  {
    "codeValue": null,
    "kind": "IR",
    "bits": 0,
    "repeatValue": false,
    "codeRaw": [
      "350",
      "600",
      " 350",
      "600",
      " 300",
      "350",
      " 600",
      "600",
      " 350",
      "250",
      " 650",
      "600",
      " 350",
      "300",
      " 650",
      "600",
      " 300",
      "350",
      " 600",
      "600",
      " 350",
      "250",
      " 650",
      "600",
      " 350",
      "300",
      " 600",
      "650",
      " 300",
      "300",
      " 650",
      "600",
      " 350"
    ],
    "codeType": -1
  }
]
```

This is my code:

```
//
// called by the main loop to check and process web requests periodically
//
void webProcessing(EthernetClient& ec)
{
  //
  //
  // As is often the case with a small embedded device, we aren't
  // necessarily fully parsing everything in a strictly conformant manner.
  // Everything works if you do things correctly; if your client sends
  // incorrectly formatted data the results are undefined (hopefully we
  // won't crash, but there's no guarantee all format errors are acted upon)
  //

  char sbuf[REQUESTBUFFERSIZE];     // XXX maybe should malloc
  struct http_rq rq(ec);

  rq.bufp = sbuf;
  rq.bufsiz = REQUESTBUFFERSIZE;
  rq.t0 = millis();                // time zero for enforcing timeouts

  if (parseHTTPheader(&rq) != -1) {
    struct urlfuncs *dtp;

    for (dtp = dt; dtp->url; dtp++) {
      if (dtp->whatop == rq.operation) {

        // if the URL pattern ends with '/' we only look at leading
        // equivalence (strncmp). This allows for extra info after
        // the endpoint. Otherwise the endpoint must match exactly.
        int n = strlen(dtp->url);
        if (dtp->url[n - 1] == '/') {
          if (strncmp(dtp->url, sbuf, n) == 0)
            break;
        }
        else {
          if (strcmp(dtp->url, sbuf) == 0)
            break;
        }
      }
    }

    const size_t bufferSize = JSON_ARRAY_SIZE(5) + 2*JSON_ARRAY_SIZE(33) + JSON_ARRAY_SIZE(99) + 2*JSON_OBJECT_SIZE(2) + 3*JSON_OBJECT_SIZE(6) + 1110;
DynamicJsonBuffer jsonBuffer(bufferSize);
 char results[1000] = {};
      char c;
      int index = 0;
      while (ec.available()) {
        c = ec.read();
        results[index] = c;
        index++;
      }
      JsonObject& json = jsonBuffer.parseObject(results);
    if (!json.success()) {
      Serial.println("JSON parsing failed!");
    } else {
        Serial.println("JSON parsing sucess!");
      }
      (*dtp->f)(ec, dtp, json);
      ++requests_processed;
    
  }
}
```
1-This code dont print the JSON parsing status, the Arduino Mega reset right before.
2-It works with a lower JSON.

If I remove:

```
char results[1000] = {};
      char c;
      int index = 0;
      while (ec.available()) {
        c = ec.read();
        results[index] = c;
        index++;
      }
```

And try to read from client,  it wont crash but the parse json fails. is the json too big?

ArduinoJson = 5.8.4
Arduino IDE = 1.8.2

bblanchon, 2017-04-18:
Hi Marcus,

The Arduino Mega has 8KB or RAM and you're using more that 10KB, just for the `JsonBuffer`.
You need to switch to a bigger MCU or use another library like [json-streaming-parser](https://github.com/squix78/json-streaming-parser).

See:

* [FAQ: Why does my device crash or reboot?](https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot/)
* [FAQ: Can I parse a JSON input that is too big to fit in memory?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

terryjmyers, 2017-04-22:
@marckaraujo
You can paste in the json here: https://bblanchon.github.io/ArduinoJson/assistant/ to see the size.

I suggest going to an ESP8266 micro controller as this would have more than enough RAM to parse that json.  I suggest searching AliExpress for "esp8266 12E"  This controller has built in wifi and you can purchase a form factor from wemos identical to the Arduino Uno form factor

closed 2017-04-28 by bblanchon

marckaraujo, 2017-05-04:
@terryjmyers , Do you know if https://www.arduino.cc/en/Main/ArduinoBoardYun solve this problem?