bblanchon/ArduinoJson #483: Compile errors using VisualStudio2015 - VisualMicro - Teensy

cfeied, 2017-04-18:
Hello,

I am seeing a large number of errors and warnings when compiling with #include ArduinoJson.h added to my code. If I ignore them all, ArduinoJson *does* compile and link and I am able to use it within my programs. Unfortunately the errors and warnings are so numerous that they prevent me from working effectively on my code.

The same thing happens when I attempt to compile JsonGeneratorExample.

The toolchain is handled by VisualMicro in Visual Studio 2015.
The target board is Teensy 3,2
I have the latest Arduino IDE and VisualMicro versions installed.

When I compile JsonGeneratorExample the output looks normal except for float range warnings. However, the VS error and warning lists are packed with errors.

I am attaching files showing the compilation output and the errorlist.  Any suggestions would be much appreciated.
[arduinojson_teensy_compilation_output.txt](https://github.com/bblanchon/ArduinoJson/files/927115/arduinojson_teensy_compilation_output.txt)
[arduinojson_teensy_error_list.txt](https://github.com/bblanchon/ArduinoJson/files/927114/arduinojson_teensy_error_list.txt)

bblanchon, 2017-04-18:
Hello @cfeied,

I think that ArduinoJson is only responsible for the following warnings:

```
warning: floating constant exceeds range of 'float' [-Woverflow]
warning: floating constant truncated to zero [-Woverflow]
```

I'll try to get rid of this.
Meanwhile, it's safe to ignore them.

I don't think the errors in `arduinojson_teensy_error_list.txt` are due to ArduinoJson, since they are not located in ArduinoJson source files.

Thank you very much for reporting this problem 👍 

Regards,
Benoit

cfeied, 2017-04-18:
Hi Benoit,

Although the errors reference other files  it seems pretty clear that the
problem is  caused by #including arduinojson in a program compiled under
visual micro for teensy. Here's why:

The errors all vanish if I don't include arduinojson. They appear in every
program in which I include arduinojson, including your examples.  They
appear whether or not I actually use arduinojson in any way: merely
#including the .h file is sufficient to trigger them.

The errors obey Coke's postulates (they  appear 100% of the time **if and
only if** arduinojson is #included) -- therefore it is highly likely that
the cause is something within AJ.

I will reach out to the folks at PJRC and Visual Micro to see whether they
can see any reason for the incompatibility, but since no other library
behaves this way I am guessing they will tell me that you are the person
best positioned to resolve the problem.

Tomorrow I will attempt to compile AJ under visualmicro with an arduino
target, to see whether the problem is associated with VMicro or with the
Teensy core per se. I'll let you know what I find.

Thanks!

-- Craig

bblanchon, 2017-04-20:
> if and **only if** arduinojson is #included

That is a very bold statement!
It would mean that you literally tested every other header on earth.

Did you try the followings?

```c++
#include <float.h>
#include <wchar.h>
#include <cmath>
#include <cstdlib>
#include <cwchar>
```

BTW, these are IntelliSense errors, right?
It seems that it's looking in the Visual C++ header instead of using the ones from Arduino.

bblanchon, 2017-04-28:
@cfeied Any progress?

cfeied, 2017-04-28:
Yes, I was forced to abandon Arduinojson, and all is well.

bblanchon, 2017-04-29:
OK, I understand.
Thanks you for reporting this issue as it allowed to fix the warnings.

kiwiwarwick, 2017-05-25:
Hello Benoit

I am having the exact same issue as cfeied.

I use the Visual Micro plug in for MS Visual Studio 2017. Compiling and uploading onto Arduino boards. I initially used the Arduino Library manager to install your library, then after getting the issues, I then downloaded a zip of the library from Github, still the errors existed.

As soon as I #include the ArduinoJson libary then I get hundreds of errors showing in the code. I take it that these are the IntelliSense errors (they are red underlines in the code). For example I get a redline under "const" in the line "const int ledPin = 13".  When I hover the mouse over the error I get the following message in a popup "variable "EMIT" is not a type name".

I have also noticed that genuine errors in the code are not pickud up by IntelliSense when this library is included.

The code does compile without any errors and does run as expected on the Arduino boards.

I'm excited to use your library, in testing with the Arduino IDE the library seems to be doing exactly what I'm after, but as with cfeied, it becomes unworkable in VisualStudio once I include your library.

Any help of thoughts would be appreciated.

Regards
Warwick

bblanchon, 2017-05-26:
Hello Warwick,

I'll try install Visual Micro on the week-end.
Meanwhile can you try to include the headers I listed in the [previous comment](https://github.com/bblanchon/ArduinoJson/issues/483#issuecomment-295651150)?

Regards,
Benoit

kiwiwarwick, 2017-05-26:
Hello Benoit

Thanks for your message. Below is some further info. I had tried adding the additional headers, but I've tried again to make sure :). (Still no luck though).

Here are some screen shots of the Visual Studio screen showing the Intellisense errors. I started with one of your examples and removed the comment lines for clarity.

It worked OK to start with (no Intellisense errors), but once I saved and compiled it a couple of times the Intellisense errors started (or maybe it was when I added an additional line with an intentional Intellisense error).

One image shows it initially, the other with the headers included. The wchar.h library is not found and the sketch fails to compile when this is included. Below is the compile error message:

JsonGeneratorExample.ino: 5:19: fatal error: wchar.h: No such file or directory
   #include <wchar.h>

I downloaded wchar.h and included it in the sketch directory, and changed the line to look locally - #include "wchar.h", this fixed the Intellisense error on that particular line, but the others still remain. But it didn't compile, this is the error:

wchar.h: 27:20: fatal error: _mingw.h: No such file or directory
   #include <_mingw.h>
   compilation terminated

I have been using your library (with VS and VM) in the last 24 hours, as a bit of a work around I comment out the #include <ArduinoJson.h> line, this only gives me a handful of errors shown (the lines associated with the library), this allows me to keep working on the code, when I compile and upload I un-comment the line.

Thanks in advance for your help on this

Regards
Warwick
[image]
[image]

kiwiwarwick, 2017-05-27:
Hello Benoit

Some additional info, when compiling for Arduino Boards (ie Mega or Nano), then there are Intellisense issues, but the ESP8266 seems fine. I notice that the headers that you suggested included are on my system in the following directory, but are not available for non ESP8266 boards

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\tools\xtensa-lx106-elf\xtensa-lx106-elf\include\c++\4.8.2\ext

Regards
Warwick

hairlesshobo, 2017-05-27:
[image]

I am encountering the exact same issue as the above users have noted using v5.10.0. In my case, I am using the latest version of VisualMicro and VS2017. Same symptoms, if I include ArduinoJson, I get hundreds of error messages. Most are 'variable "EMIT" is not a type name' but I also noticed that there are a handful of other errors as well. When I comment out the #include <ArduinoJson.h>, I get 0 errors in my solution.  I have attached a txt file containing the full list of errors, sorted by file name in case this helps at all.
[Error List.txt](https://github.com/bblanchon/ArduinoJson/files/1033318/Error.List.txt)

For good measure, I confirmed this issue happens whether I include a library installed to the sketch folder, a full source tree included in my project, or either of the single file releases that is available to download on GitHub under releases. 

I have also confirmed this same behavior with v5.8.4, but I was unable to replicate the issue with 5.8.3. I decided to clone the repo and worked my way up, commit by commit until I was able to determine which commit is what triggered the issue. From the best I can tell, the issue first appeared in commit c4567bac1891e40669eab4b128737760da6ea823. 

If time permits this weekend, I will attempt to dig a bit deeper and see if I can figure out exactly what change caused it, but it may take a while since I am unfamiliar with the source of ArduinoJson.

Thanks in advance!

hairlesshobo, 2017-05-27:
I did some further digging, and I found that the problem is caused by the visual studio specific code in include/ArduinoJson/Polyfills/math.hpp. Apparently calling #include <limits> causes the environment to use a copy of the limits header that really upsets things in this context (I wish I could elaborate more, but I really have no idea exactly what is happening here vs what should be happening here). 

I am afraid, at least at this time of night, that figuring out a fix for it is beyond my mental abilities. Maybe later this weekend I will be able to figure something out.

bblanchon, 2017-05-27:
Thank you very much @hairlesshobo for pointing out the commit.

The guilty line is [Polyfills/math.hpp:14](https://github.com/bblanchon/ArduinoJson/blob/master/src/ArduinoJson/Polyfills/math.hpp#L14):

```c++
// If Visual Studo
#if defined(_MSC_VER)

#include <float.h>
#include <limits> // <-- this header is causing the troubles
```

This is because `_MSC_VER` is defined by the IntelliSense compiler.
I don't know if the authors of Visual Micro can fix that.

I'm looking for a workaround, hold on.

bblanchon, 2017-05-27:
Please checkout [branch issue483](https://github.com/bblanchon/ArduinoJson/tree/issue483)

hairlesshobo, 2017-05-27:
That did the trick, @bblanchon! No more intellisense errors and the project compiles just fine.

Thanks for the fast turnaround!!

bblanchon, 2017-05-28:
Thank you Steve, you did most of the work.

closed 2017-05-28 by bblanchon

mo2menelzeiny, 2017-06-12:
I'm still experiencing the same errors on Visual-Micro, I tried a new empty Arduino Project with only ArduinoJson included and around 1108 errors are produced.