bblanchon/ArduinoJson #485: How to convert a JSON object to String or Char*

duythang, 2017-04-19:
Thank you so much for your nice work.

I am trying to use your lib to send data over internet. My problem is that
i want to convert a JSON object to String or Char*. However, i don't know how it does.
Hope you can help me.

tarzan115, 2017-04-19:
```c++
#include <ArduinoJson>
StaticJsonBuffer<SERVER_DATA_JSON_SIZE> jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(jsonFromServer);

const char* yourstring = root["your object"];
```
or
```c++
#include <ArduinoJson>
StaticJsonBuffer<SERVER_DATA_JSON_SIZE> jsonBuffer;
JsonObject& root = jsonBuffer.parseObject(jsonFromServer);

struct DataFromServer
{
     const char* yourString;
     // something
}

DataFromServer data;
data.yourString = root["your object"];
```

I hope this will help you do that 😄

duythang, 2017-04-19:
@tarzan115: Thank you, Luong. 
Your code is only for parsing a JSON object of the received msg. I know about this. 
However, in my case, i want to convert a JSON object of the msg to String before i send this msg to internet.

tarzan115, 2017-04-19:
here you are 😃 
```c
#include <ArduinoJson>
StaticJsonBuffer<CLIENT_DATA_JSON_SIZE> jsonBuffer;
JsonObject& root  = jsonBuffer.createObject();
//do something to generate your json

char jsonChar[100];
root.printTo((char*)jsonChar, root.measureLength() + 1);
```

duythang, 2017-04-19:
@tarzan115 : Thank you so much. I can run well :)

bblanchon, 2017-04-19:
@tarzan115 There is a risk of a buffer overrun in your last sample.

Here is the correct code:

```c++
char jsonChar[100];
root.printTo(jsonChar);
// or root.printTo(jsonChar, sizeof(jsonChar)); if you want to be more explicit
```

Note that you can also write to a String:

```c++
String jsonStr;
root.printTo(jsonStr);
```

See:

* [API Reference: JsonObject::printTo()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/printto/)

closed 2017-04-19 by bblanchon

duythang, 2017-04-19:
@bblanchon : Thank you so much

sanfx, 2017-11-23:
Well thats quite embarrassing,  I am also in the same situation but but trying the same example given above doesnt work, 
```

JsonObject& prepareResponse(JsonBuffer& jsonBuffer) {
  JsonObject& root = jsonBuffer.createObject();
  JsonArray& tempValues = root.createNestedArray("temperature");
  tempValues.add(t);
  JsonArray& humiValues = root.createNestedArray("humidity");
  humiValues.add(h);
  JsonArray& dewpValues = root.createNestedArray("dewpoint");
  dewpValues.add(pfDew);
  JsonArray& EsPvValues = root.createNestedArray("systemv");
  EsPvValues.add(pfVcc / 1000, 3);
  return root;
}

void writeResponse(WiFiClient& clnt, JsonObject& json) {
  clnt.println("HTTP/1.1 200 OK");
  clnt.println("Content-Type: application/json");
  clnt.println("Access-Control-Allow-Origin: *");
  clnt.println("Connection: close");
  clnt.println();
  json.prettyPrintTo(clnt);
}

void outputJson() {
  StaticJsonBuffer<500> jsonBuffer;
  JsonObject& json = prepareResponse(jsonBuffer);
  writeResponse(client, json);
  //  server.send ( 200, "text/json", json  );
}

```

I am using ```ESP8266WebServer.h```  to create server object of ```WifiServer```.

rizban114406, 2018-03-12:
#include <ArduinoJson.h>
int8_t answer;
void setup() {
Serial.begin(9600);
setNetworkGSM();

}

void loop() {
char body[200];
char response[100];
DynamicJsonBuffer jBuffer;
JsonObject& root = jBuffer.createObject();
JsonObject& root2 = jBuffer.createObject();
JsonArray& data = root.createNestedArray("data");
root2["userid"] = 1;
root2["trip_reference"] = 12434243;
root2["latitude"] = "23.7907832";
root2["longitude"] = "90.4090195";
root2["date"] = "20160510";
root2["time"] = "12:10:02";
root2["speed"] = "100";
root2["distance"] = "100";
data.add(root2);
root.printTo(body);
//Serial.println(body);

doHttpRequests(body,response);
Serial.println(response);
}
void setNetworkGSM()
{
delay(10000);
while (sendATcommand("AT+CREG?", "+CREG: 0,1", 2000) == 0);
sendATcommand("AT+SAPBR=3,1,"Contype","GPRS"", "OK", 2000);
sendATcommand("AT+SAPBR=3,1,"APN","internet"", "OK", 2000);
while (sendATcommand("AT+SAPBR=1,1", "OK", 20000) == 0)
{
delay(5000);
}
}
void doHttpRequests(char* body, char* response)
{
answer = sendATcommand("AT+HTTPINIT", "OK", 10000);
//String data = body;
if (answer == 1)
{
answer = sendATcommand("AT+HTTPPARA="CID",1", "OK", 5000);
if (answer == 1)
{
char aux_str[300];
sprintf(aux_str, "AT+HTTPPARA="URL","http://Server_IP_Address/carcopolo_arduino/Json_index.php?tag=%s&data=%s","insertdevicedata",body);
Serial.print(aux_str);
answer = sendATcommand(""", "OK", 10000);
if (answer == 1)
{
answer = sendATcommand("AT+HTTPACTION=0", "+HTTPACTION: 0,200", 15000);
if (answer == 1)
{
answer = sendATcommand("AT+HTTPREAD", "+HTTPREAD:", 35000);

      if(answer == 1)
      {
        readResponse(response,10000);
      }
    }
  }
}
}
sendATcommand("AT+HTTPTERM", "OK", 5000);
}

Here is my code. I have made a json object and copied it to a character array named body to send it to the server via get request. I have checked the serial monitor and it prints perfectly.
AT+HTTPPARA="URL","
http://Server_IP_Address/carcopolo_arduino/Json_index.php?tag=insertdevicedata&data={
"data":[{"userid":1,"trip_reference":12434243,"latitude":"23.7907832","longitude":"90.4090195","date":"20160510","time":"12:10:02","speed":"100","distance":"100"}]}"

But I am getting an Error response. I have copied the url and tested it in a browser and it worked fine. Can you please tell me what mistake i have done that i am getting this error??? thanks in advance.