bblanchon/ArduinoJson #486: Json Parsing fails on Wemos D1R2

zhajar, 2017-04-19:
Hello,

I'm a newbie here but have a decent background with other database languages. I've been trying to use your code to parse a Json response from a weather service. The Json Parser is not returning results for reasons I cannot understand/explain (probably my fault).

The ultimate objective is to parse total rainfall and use this information to offset the irrigation cycle (irrigation system is currently running on an arduino).

Hope you can help
Thanks

**_Platform:_** WeMos D1R2

**_Code:_**
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266HTTPClient.h>
#define USE_SERIAL Serial
#include <ArduinoJson.h>
DynamicJsonBuffer jsonBuffer;
ESP8266WiFiMulti WiFiMulti;

void setup() {
    DynamicJsonBuffer jsonBuffer;
    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

    for(uint8_t t = 4; t > 0; t--) {
        USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
        USE_SERIAL.flush();
        delay(1000);
    }

    WiFiMulti.addAP("********", "*******");

}

void loop() {
    // wait for WiFi connection
    if((WiFiMulti.run() == WL_CONNECTED)) {

        HTTPClient http;

        USE_SERIAL.print("[HTTP] begin...\n");
        // configure traged server and url
        //http.begin("https://192.168.1.12/test.html", "7a 9c f4 db 40 d3 62 5a 6e 21 bc 5c cc 66 c8 3e a1 45 59 38"); //HTTPS
        http.begin("http://api.wunderground.com/api/***********/conditions/q/**/******.json"); //HTTP

        USE_SERIAL.print("[HTTP] GET...\n");
        // start connection and send HTTP header
        int httpCode = http.GET();

        // httpCode will be negative on error
        if(httpCode > 0) {
            // HTTP header has been send and Server response header has been handled
            USE_SERIAL.printf("[HTTP] GET... code: %d\n", httpCode);

            // file found at server
            if(httpCode == HTTP_CODE_OK) {
                String payload = http.getString();
                USE_SERIAL.println(payload);
                JsonObject& root = jsonBuffer.parseObject(payload);
                if (!root.success()) {
                Serial.println("parseObject() failed");
                }
                String temp = root["temp_f"];
                USE_SERIAL.println(temp);
            }
        } else {
            USE_SERIAL.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
        }
        http.end();
    }
    delay(120000);
}

**_JSON being parsed:_**

{
  "response": {
  "version":"0.1",
  "termsofService":"http://www.wunderground.com/weather/api/d/terms.html",
  "features": {
  "conditions": 1
  }
	}
  ,	"current_observation": {
		"image": {
		"url":"http://icons.wxug.com/graphics/wu2/logo_130x80.png",
		"title":"Weather Underground",
		"link":"http://www.wunderground.com"
		},
		"display_location": {
		"full":"Sunrise, FL",
		"city":"Sunrise",
		"state":"FL",
		"state_name":"Florida",
		"country":"US",
		"country_iso3166":"US",
		"zip":"33323",
		"magic":"2",
		"wmo":"99999",
		"latitude":"26.14999962",
		"longitude":"-80.33000183",
		"elevation":"3.0"
		},
		"observation_location": {
		"full":"Plantation, FL just off Commadore, Plantation, Florida",
		"city":"Plantation, FL just off Commadore, Plantation",
		"state":"Florida",
		"country":"US",
		"country_iso3166":"US",
		"latitude":"26.123884",
		"longitude":"-80.329079",
		"elevation":"19 ft"
		},
		"estimated": {
		},
		"station_id":"KFLPLANT24",
		"observation_time":"Last Updated on April 18, 11:38 PM EDT",
		"observation_time_rfc822":"Tue, 18 Apr 2017 23:38:54 -0400",
		"observation_epoch":"1492573134",
		"local_time_rfc822":"Tue, 18 Apr 2017 23:40:06 -0400",
		"local_epoch":"1492573206",
		"local_tz_short":"EDT",
		"local_tz_long":"America/New_York",
		"local_tz_offset":"-0400",
		"weather":"Mostly Cloudy",
		"temperature_string":"74.8 F (23.8 C)",
		"temp_f":74.8,
		"temp_c":23.8,
		"relative_humidity":"74%",
		"wind_string":"From the NE at 1.0 MPH Gusting to 3.0 MPH",
		"wind_dir":"NE",
		"wind_degrees":44,
		"wind_mph":1.0,
		"wind_gust_mph":"3.0",
		"wind_kph":1.6,
		"wind_gust_kph":"4.8",
		"pressure_mb":"1021",
		"pressure_in":"30.15",
		"pressure_trend":"+",
		"dewpoint_string":"66 F (19 C)",
		"dewpoint_f":66,
		"dewpoint_c":19,
		"heat_index_string":"NA",
		"heat_index_f":"NA",
		"heat_index_c":"NA",
		"windchill_string":"NA",
		"windchill_f":"NA",
		"windchill_c":"NA",
		"feelslike_string":"74.8 F (23.8 C)",
		"feelslike_f":"74.8",
		"feelslike_c":"23.8",
		"visibility_mi":"10.0",
		"visibility_km":"16.1",
		"solarradiation":"--",
		"UV":"0","precip_1hr_string":"0.00 in ( 0 mm)",
		"precip_1hr_in":"0.00",
		"precip_1hr_metric":" 0",
		"precip_today_string":"0.00 in (0 mm)",
		"precip_today_in":"0.00",
		"precip_today_metric":"0",
		"icon":"mostlycloudy",
		"icon_url":"http://icons.wxug.com/i/c/k/nt_mostlycloudy.gif",
		"forecast_url":"http://www.wunderground.com/US/FL/Sunrise.html",
		"history_url":"http://www.wunderground.com/weatherstation/WXDailyHistory.asp?ID=KFLPLANT24",
		"ob_url":"http://www.wunderground.com/cgi-bin/findweather/getForecast?query=26.123884,-80.329079",
		"nowcast":""
	}
}

bblanchon, 2017-04-19:
Hi @zhajar,
Did you try to use the code generated by the [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)?

zhajar, 2017-04-20:
This is an absolutely awesome tool! Thank you so much.
The details of the code are way over my level of understanding but it works. Thanks again!

closed 2017-04-20 by zhajar