bblanchon/ArduinoJson #487: Unable to use  F() macro on any object reference than the first

terryjmyers, 2017-04-21:
Trying to use the F() macro like this does NOT work:

`
aVariable = json[F("someObject")][F("someMember")];`
or
`aVariable = json[F("someObject")][F("someMember")][F("anotherMember")];`

However...You can F() the first object call.  So for example this would work:
```
//this works
String temp = F("someMember");
aVariable = json[F("someObject")][temp];
```
or 
```
//this works
String temp = F("someMember");
String temp2 = F("anotherMember");
aVariable = json[F("someObject")][temp][temp2];
```

bblanchon, 2017-04-22:
Hello Terry,

I'm not able to reproduce this bug.

1. Can you provide a minimal program to reproduce it?
2. What platform are you working on?
3. Do you have a compilation error? if so please copy and paste the compiler output.

Regards,
Benoit

terryjmyers, 2017-04-22:
Thanks for taking this up. As requested:

I'm using Visual Studio with the vmicro plugin.  Arduino IDE v1.8.2 is installed providing the toolchain.  I'm downloading to an ESP8266.  The bug is reproducable with any of the boards selected, but let's say specifically the "NodeMCU 1.0 (ESP-12E Module)"

I programmed a minimal sketch as requested.  Here is the full VS project:
https://www.dropbox.com/s/6b0829t9x6izhg8/Sketch2.7z?dl=0

Alternatively here is my sketch:
```
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson
String jsonToDeSerialize = "\
{\n\
   \"results\" : [\n\
      {\n\
         \"geometry\" : {\n\
            \"location\" : {\n\
				\"lat\" : 40.141849,\n\
				\"lng\" : -75.86452799999999\n\
            }\n\
         }\n\
      }\n\
   ]\n\
}"; //adappted from a stripped down version of maps.google.api: http://maps.googleapis.com/maps/api/geocode/json?address={Fill an address in with plus signs}

void setup() {
	Serial.begin(250000);
	Serial.println("json to deserialize:");
	Serial.println(jsonToDeSerialize);

	DynamicJsonBuffer jsonBuffer(1000);

	JsonObject& json = jsonBuffer.parseObject(jsonToDeSerialize);
	if (json.success()) {

		//This doesn't compile:
			//float lat = json[F("results")][0][F("geometry")][F("location")][F("lat")];

		//But this does:
			String geometry = F("geometry");
			String location = F("location");
			String lat1 = F("lat");
			float lat = json[F("results")][0][geometry][location][lat1];
		
			
		Serial.println();
		Serial.print("Latitude Result: ");  Serial.println(lat);
	}
	else {
		Serial.println("no....no...");
	}
}

void loop() {

}
```

Here is the compiler output which was a bit too unwieldy to past here:
regular: https://www.dropbox.com/s/cmsq3i3ptlemls4/FMacroBuildDebugWindow.txt?dl=0
Verbose: https://www.dropbox.com/s/ie9fczurkm1991y/FMacroBuildDebugWindowVerbose.txt?dl=0

closed 2017-04-23 by terryjmyers