bblanchon/ArduinoJson #488: Cannot parse 5 day openweathermap json

AcuarioCat, 2017-04-23:
Hi,
I am trying to process the openweathermap 5 day 3 hour forecast. I used the assistant to generate the code but the json appears not to be parsed. I have attached the generated code (with some prettyprint added) and the original json. I also noticed that only code for the the first 3 elements of the array is generated (this isn't a problem for me as I only need the first 2). As the code is basically repeated I thought it was just truncated.
I'm using an ESP8266 if that makes any difference.

[jsonproblem.txt](https://github.com/bblanchon/ArduinoJson/files/949642/jsonproblem.txt)
[jsonfile.txt](https://github.com/bblanchon/ArduinoJson/files/949643/jsonfile.txt)

bblanchon, 2017-04-23:
Hi @AcuarioCat,

The [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/) says you need 27KB, just for the `JsonBuffer`.
That's very big, even for an ESP8266.
I bet that you are running out of memory.
Did you try with a smaller input?

AcuarioCat, 2017-04-23:
With the current forecast it works fine, it's the 5 day one that fails. Unfortunately the data returned from the api call to openweathermap is not under my control.
Any ideas?
Thanks

bblanchon, 2017-04-28:
I'm afraid you'll not be able to use ArduinoJson in this case.
You should try with [json-streaming-parser](https://github.com/squix78/json-streaming-parser).

Have a look at:

* [FAQ: Can I parse a JSON input that is too big to fit in memory?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

closed 2017-05-07 by bblanchon