bblanchon/ArduinoJson #489: ESP8266 strange behavier

merlokk, 2017-04-26:
Platform: arduino + ESP8266

this code works ok:

```
  char jsonMem[JSON_MEM_BUFFER_LEN] = {0};

  StaticJsonBuffer<JSON_OBJ_BUFFER_LEN> jsonBuffer;
  JsonObject &root = &jsonBuffer.parseObject(String(jsonMem));
  if (!root.success()){
    return
  };
  root.set(paramName, paramValue);
  root.printTo(&jsonMem[0], JSON_MEM_BUFFER_LEN);
```

but if i change `JsonObject &root = &jsonBuffer.parseObject(String(jsonMem));` to `JsonObject &root = &jsonBuffer.parseObject(jsonMem);` it will corrupt `jsonMem` contents

as I see this code:
```
  StaticJsonBuffer<JSON_OBJ_BUFFER_LEN> jsonBuffer;
  JsonObject &root = &jsonBuffer.parseObject(jsonMem);
  root.printTo(&jsonMem[0], JSON_MEM_BUFFER_LEN);
```
corrupts `jsonMem`

merlokk, 2017-04-26:
after some investigation i see that code deletes <">, <:>, <"> or adds some <\> near <">

merlokk, 2017-04-26:
ps. code here https://github.com/merlokk/SmartHome/blob/master/lib/xparam.h

bblanchon, 2017-04-27:
Hi Oleg,

This is intentionnal but it's not propery documented.
I added the following page to the FAQ:

* [FAQ: Why is the input modified?](https://bblanchon.github.io/ArduinoJson/faq/why-is-the-input-modifed/)

Regards,
Benoit

merlokk, 2017-04-27:
but it modifies not only escape characters. it modifies <"> chfracters. It made the string fully corrupted.

bblanchon, 2017-04-27:
If this is a problem for you, just cast your input to a `const char*`.

merlokk, 2017-04-27:
works(

merlokk, 2017-04-27:
https://github.com/merlokk/SmartHome/commit/0411070981f3ef435757d1c3e463511f10ed8a97
https://github.com/merlokk/SmartHome/commit/0411070981f3ef435757d1c3e463511f10ed8a97

closed 2017-04-27 by merlokk