bblanchon/ArduinoJson #491: ESP32 Support

anteph, 2017-04-27:
Hi! sorry for opening an issue but didn't find any email for contacting you. 

Just tested some functionalities of the library with the ESP32 and it is working fine. Since the support isn't mentioned yet in the supported hardware list, thought it would be useful to share, so others know they can use it.

I've tested the encoding of a JSON message and the parsing of a JSON message with success. I've written two blog posts about it, please let me know if you want me to share here. I haven't done so because I don't know if it is interpreted as spam.

Best regards, 
Nuno Santos

bblanchon, 2017-04-28:
Hi Nuno,

Thank you very much for your feedback.

No it's not considered as spam. *Au contraire!*
I wish I could receive more messages like this 😉 

Please add the link to your blog post in the wiki:
https://github.com/bblanchon/ArduinoJson/wiki/Projects-using-ArduinoJson

Regards,
Benoit

closed 2017-04-28 by bblanchon

anteph, 2017-04-28:
Hi Benoit,

Just added the ESP32 links to the Wiki, and some on the ESP8266 too. 

Thanks for your great work on this awesome library, I will definitely keep using it in my projects and recommend to others :)

Regards,
Nuno Santos