bblanchon/ArduinoJson #493: Issue when trying to add an Object to a nested array

Alexpelillo, 2017-05-01:
Hi BBlanchon, I can't thank you enough for making this library, but as I'm trying to make a system that loads a structure from memory filesystem (Developing on esp8266 NodeMcu with Arduino IDE), the only resort that i'm encountering right now is this one:
The structure is like this:
{"Devices":[
                  {"Device":"Circutor CVM-mini",
                   "Vars":[
                             {"VrName":"VRMS",
                                "VrAddr":"20"}]}}]
With more than 1 device and with more than 1 variable each. I am able to load and store from memory, and delete variables too with array.removeAt()... (will try further to delete devices) But I can't create variables inside the vars array. Tried with varArray.createNestedObject() and then filling the Object with data, and tried with first creating the Object and then varArray.add(Jobject) and I keep getting issues with the ESP8266 rebooting and the stack is like this:

3fff0864:  00000014 00000014 40206ddc 3ffef0c0  
3fff0874:  00000000 3ffef0c0 4020b469 3ffe89e8  
3fff0884:  3fff0890 00000190 00000008 3fff0880  
3fff0894:  00000000 736a2e69 00006e6f 3ffedd5c  
3fff08a4:  00000000 4021eaa8 00000000 3ffefa84  
3fff08b4:  00000013 00000013 4010020c 3ffe84c0  
3fff08c4:  3ffef0c0 3fff0920 4010068c 00000031  
3fff08d4:  0000000a 3fff2fec 4020f95c 3ffe84c0  
3fff08e4:  3ffef0c0 3fff2fec 40206953 3ffefa84  
3fff08f4:  00000012 00000012 3ffefa84 00000010  
3fff0904:  3fff3034 00000000 4010053d 00000008  
3fff0914:  00000000 00000020 40101c72 3ffe98e8  
3fff0924:  4010191d 00000002 40101c72 3ffeda60  
3fff0934:  40101806 00000001 00000000 00000000  
3fff0944:  401015d3 00000000 3ffeda30 00000005  
3fff0954:  00000000 00000020 40101c72 3ffe98e5  
3fff0964:  40104f4f 3ffed370 3ffefa84 00000010  
3fff0974:  3fff0a4c 00000000 4010053d 00000010  
3fff0984:  3fff0a40 00000000 4010053d 3ffea184  
3fff0994:  00000000 00000000 3fff3520 00000010  
3fff09a4:  3fff0a4c 3fff0a4c 4020f24c 3ffe9518  
3fff09b4:  00000000 3fff0a4c 4020f531 3ffe9518  
3fff09c4:  00000000 3fff0a4c 00000001 00000001  
3fff09d4:  3fff0a4c 3fff0a40 4020f5b4 3ffe0033  
3fff09e4:  3fff0a90 3fff0a40 4020f445 3fff31f1  
3fff09f4:  3ffe9608 3fff0a40 4020da7b 3ffefa84  
3fff0a04:  00000451 00000451 4010020c 00000007  
3fff0a14:  3fff0a90 3fff0a40 4010068c 00000000  
3fff0a24:  00000000 00000000 4020f1f0 00000007  
3fff0a34:  3fff0a90 3ffef0c0 3ffefa84 00000020  
3fff0a44:  3fff0ad0 00000000 4010053d 3ffefa84  
3fff0a54:  000003f5 000003f5 3ffefa84 00000020  
3fff0a64:  3fff0af0 00000000 4020d53c 00000020  
3fff0a74:  3fff0ad0 3fff0ad0 4020f24c 00000001  
3fff0a84:  00000000 3fff0890 4020f29b 3ffefa84  
3fff0a94:  000003d4 000003d4 4020d53c 00000001  
3fff0aa4:  00000001 3fff25e4 4020b910 00000000  
3fff0ab4:  3fffdad0 3fff25e4 4020d532 3fff25e4  
3fff0ac4:  3ffef100 3fff25e4 4020d56e 00000000  
3fff0ad4:  00000000 00000000 4020f3fc 3fff25e4  
3fff0ae4:  3ffef100 3ffef0c0 4020d601 3fff31bc  
3fff0af4:  0000001f 00000013 4020c588 3ffef100  
3fff0b04:  feefeffe 3ffefb70 00000001 00000001  
3fff0b14:  4020cb44 0000000f 3ffefb70 00000000  
3fff0b24:  00000000 3ffef0c0 3ffefb64 00000001  
3fff0b34:  3ffef0e4 3ffef0c0 4020d7e3 3ffe8a30  
3fff0b44:  00000000 000003e8 feefeffe 00000000  
3fff0b54:  3fff3234 feefeffe feefeffe 3fffdad0  
3fff0b64:  00000000 3ffefb5c 40206b98 3fffdad0  
3fff0b74:  00000000 3ffefb5c 4020fa70 feefeffe  
3fff0b84:  feefeffe 3ffefb70 40100718 feefeffe  
 with exceptions (2) and (9) (if I try the first method or the second one) is it memory issue?
I have been optimising the code so there's only the Jobject root only one time opened and from there I access the subObjects and Arrays of Devices and Variables like this: root["Device"][number_of_device]["Vars"][variable]...

Alexpelillo, 2017-05-02:
If I instead choose by JsonObject& nested = variablesArray.createNestedObject(); I get a wdt reset error with no stack so I can't debug it and I can't then put the data in the JsonObject& nested because of wdt reset error.

bblanchon, 2017-05-03:
Hello @Alexpelillo,

Can you share the source code?
Can you decode the stack trace using [this awesome tool from @me-no-dev](https://github.com/me-no-dev/EspExceptionDecoder)?

Regards,
Benoit

Alexpelillo, 2017-05-03:
String json_s = ",{\"Data\":\"0\",\"VrName\":\"" +(String)server.arg("vn") +"\",\"VrAddr\":\"" +(String)server.arg("va") +"\",\"VTyp\":\"" +(String)server.arg("vt") +"\",\"Dtyp\":\"" +(String)server.arg("dt") +"\",\"Xtra1\":\"" + (String)server.arg("x1") +"\",\"Xtra2\":\"" + (String)server.arg("x2") + "\"}";

        DynamicJsonBuffer tempBuffer;
        JsonObject& jobject= tempBuffer.parseObject(json_s);
        delay(10);
        JsonObject& config_devices = loadConfig("config_devices");
        JsonArray& variablesArray=config_devices["Devices"][dispositivo_actual]["Vars"];
        variablesArray.prettyPrintTo(Serial);
        // All well until here?
        variablesArray.add(jobject);
        Serial.println("Variable created (for debugging)");
        saveConfig(config_devices, "config_devices");
        delay(20);

and the functions loadConfig, saveConfig:

```
bool saveConfig(JsonObject& config_to_save, String filename) {
  //DynamicJsonBuffer jsonBuffer;
  JsonObject& json = config_to_save; //redundante??
  Serial.println("-- Guardando JSON en memoria... --");
  File configFile = SPIFFS.open("/" + filename + ".json", "w");
  if (!configFile) {
    Serial.println("Failed to open config file for writing");
    return false;
  }

  json.printTo(configFile);
  return true;
}

JsonObject& loadConfig(String filename) { // "config_devices", "config_wifi",...
  File rootConfigFile = SPIFFS.open("/" + filename + ".json", "r");
  if (!rootConfigFile) {
    Serial.println("Failed to open config file: /" + filename + ".json");
  }

  size_t size = rootConfigFile.size();
  if (size > MAX_FILE_SIZE) {
    Serial.println("Config file size /" + filename + ".json is too large");
  }

  std::unique_ptr<char[]> buf(new char[size]);  // Allocate a buffer to store contents of the file. --> buf

  rootConfigFile.readBytes(buf.get(), size); // Leer bytes del buffer buf

  DynamicJsonBuffer jsonBuffer; // Creamos un buffer JSON estat
  JsonObject& config_root = jsonBuffer.parseObject(buf.get()); // parseamos el contenido de buf --> config_root

  if (!config_root.success()) {
    Serial.println("Failed to parse config file /" + filename + ".json");
  }
  return config_root;

}

```
aaaand the json file (short one) (Object with nested array, which has nested array with nested Objects)
```
{
  "Rubbish": "empty",
  "Devices": [
    {
      "Device": "Circutor CVM-mini",
      "DevAddr": "1",
      "Vars": [
        {
          "Data": "0",
          "VrName": "VRMS",
          "VrAddr": "20",
          "VTyp": "LE",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "5"
        },
        {
          "Data": "0",
          "VrName": "IRMS",
          "VrAddr": "22",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "NaN",
          "Xtra2": "30"
        },
        {
          "Data": "0",
          "VrName": "VPICO",
          "VrAddr": "24",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "15"
        },
        {
          "Data": "0",
          "VrName": "POWER",
          "VrAddr": "24",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "15"
        }
      ]
    }
   
  ]
}
```

Alexpelillo, 2017-05-03:
Sorry for comments being in Spanish, and for the Json, it would have more devices and more variables in each device, but that will do. All the functions like loading the Json, modifying it, saving it, deleting a variable or deleting a device ( like variablesArray.removeAt() ), are working in some way (with delays so the ESP8266 does not reset, with rubbish at starting of the JSON so it reads it corrupt but it does not do a thing with it...)

bblanchon, 2017-05-03:
I'm sorry, there is too much code 😢 
Can you write a minimalist program to reproduce the issue?
In other words, what are the minimal steps to reproduce the crash?
Also, please make sure you use the latest version of ArduinoJson.

Alexpelillo, 2017-05-03:
If you can please make an example to add a variable to my JSON... You don't have to store it in memory just get the JSON to a progmem string, parse it and when you have it try to add a variable or add a device or just add a blank object. Just focus please on the JSON:
```
{
  "Rubbish": "empty",
  "Devices": [
    {
      "Device": "Circutor CVM-mini",
      "DevAddr": "1",
      "Vars": [
        {
          "Data": "0",
          "VrName": "VRMS",
          "VrAddr": "20",
          "VTyp": "LE",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "5"
        },
        {
          "Data": "0",
          "VrName": "IRMS",
          "VrAddr": "22",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "NaN",
          "Xtra2": "30"
        },
        {
          "Data": "0",
          "VrName": "VPICO",
          "VrAddr": "24",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "15"
        },
        {
          "Data": "0",
          "VrName": "POWER",
          "VrAddr": "24",
          "VTyp": "L",
          "Dtyp": "int16",
          "Xtra1": "Sum",
          "Xtra2": "15"
        }
      ]
    }
   
  ]
}
```

Alexpelillo, 2017-05-03:
Found out also that if I load the pointer to variablesArray, I can print it well so it's not an issue of getting the right pointer: 

```
        JsonObject& config_devices = loadConfig("config_devices");
        JsonArray& variablesArray=config_devices["Devices"][dispositivo_actual]["Vars"];
        //JsonObject& nested=variablesArray.createNestedObject();
        Serial.println("Variable Creada!!!");
        //variablesArray.add(otherobj);
        delay(50);
        variablesArray.prettyPrintTo(Serial);
        delay(20);
```
I get by serial:
```
[
  {
    "Data": "0",
    "VrName": "VRMS",
    "VrAddr": "20",
    "VTyp": "LE",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "5"
  },
  {
    "Data": "0",
    "VrName": "IRMS",
    "VrAddr": "22",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "NaN",
    "Xtra2": "30"
  },
  {
    "Data": "0",
    "VrName": "VPICO",
    "VrAddr": "24",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "15"
  },
  {
    "Data": "0",
    "VrName": "POWER",
    "VrAddr": "24",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "15"
  },
  {
    "Data": "0",
    "VrName": "FP",
    "VrAddr": "24",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "15"
  },
  {
    "Data": "0",
    "VrName": "DAY",
    "VrAddr": "10",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "11"
  },
  {
    "Data": "0",
    "VrName": "PRICE",
    "VrAddr": "32",
    "VTyp": "L",
    "Dtyp": "int16",
    "Xtra1": "Sum",
    "Xtra2": "5"
  },
  {
    "Data": "0",
    "VrName": "INIT",
    "VrAddr": "02",
    "VTyp": "LE",
    "Dtyp": "0/1",
    "Xtra1": "NaN",
    "Xtra2": "10"
  }
]
```
But if I uncomment the createnested.... even if it's blank, I get exception (9),
` LoadStoreAlignmentCause	Load or store to an unaligned address` 
and if I uncomment the .add (with the object created before in the code), i get exception (2)
	`InstructionFetchErrorCause	Processor internal physical address or data error during instruction fetch`

bblanchon, 2017-05-03:
Well that's very interesting 👍 
Can you get the stack trace, so I can see where the exception happens?

Then, please try to add the following line before including ArduinoJson:

```c++
#define ARDUINOJSON_ENABLE_ALIGNMENT 1
#include <ArduinoJson.h>
```

It enables aligned allocation.
This should already be set by default, but it's worth trying.

Alexpelillo, 2017-05-03:
I added that define, and with CreateNestedObject inside the array the stack trace is the following:
```
0x40207188: ArduinoJson::Internals::ListConstIterator  ArduinoJson::JsonObject::findKey (char const*) const at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1176
0x40206db8: ArduinoJson::JsonBuffer::createObject() at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1176
0x4020bf2a: ArduinoJson::JsonArray::createNestedObject() at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonArrayImpl.hpp line 25
:  (inlined by) operator() at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 615
0x4010020c: _umm_free at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x402121c0: String::~String() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x4010053d: _umm_realloc at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4020fc88: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4021221c: String::changeBuffer(unsigned int) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x4020fc88: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020c830: _M_invoke at c:\users\alex.desktop-pp1qg6i\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4020fc7e: std::function ::operator()() const at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020fcba: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x402123cc: String::String(String const&) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x4020fd4d: ESP8266WebServer::_handleRequest() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020ecd8: WiFiClient::~WiFiClient() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020f290: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020ff2f: ESP8266WebServer::handleClient() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40206e38: loop at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1176
0x40212ab8: loop_wrapper at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/cont.S line 109

```

Alexpelillo, 2017-05-03:
With variable.add(jobject) the stack trace is the following (Exception 9 too):
```
0x4010020c: _umm_free at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x402070d0: ArduinoJson::Internals::List ::add() at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1177
0x40207104: add_impl  at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1177
:  (inlined by) ArduinoJson::TypeTraits::EnableIf::value, bool>::type ArduinoJson::JsonArray::add (ArduinoJson::JsonObject const&) at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/JsonArray.hpp line 66
0x4020c049: IndentedPrint at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/IndentedPrint.hpp line 22
:  (inlined by) prettyPrintTo  at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp line 88
:  (inlined by) operator() at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 635
0x4010053d: _umm_realloc at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1491
:  (inlined by) realloc at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266\umm_malloc/umm_malloc.c line 1709
0x4020fcd4: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40212268: String::changeBuffer(unsigned int) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x402122b7: String::reserve(unsigned int) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x4020fcd4: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020c87c: _M_invoke at c:\users\alex.desktop-pp1qg6i\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4020fcca: std::function ::operator()() const at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020fd06: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40212418: String::String(String const&) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/WString.cpp line 519
0x4020fd99: ESP8266WebServer::_handleRequest() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020ed24: WiFiClient::~WiFiClient() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020f2dc: FunctionRequestHandler::canHandle(HTTPMethod, String) at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x4020ff7b: ESP8266WebServer::handleClient() at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40206e38: loop at C:\Users\ALEX.DESKTOP-PP1QG6I\Documents\Arduino\ESP8266-apr26-webserver-spiffs-jsonv2/ESP8266-apr26-webserver-spiffs-jsonv2.ino line 1177
0x40212b04: loop_wrapper at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\ALEX.DESKTOP-PP1QG6I\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\cores\esp8266/cont.S line 109

```

bblanchon, 2017-05-03:
I'm sorry but I don't manage to reproduce the issue on my ESP8266.
Can you provide a complete sketch that I can run?

I'm using [ESP8266 core for Arduino](https://github.com/esp8266/Arduino), maybe the problem is related to NodeMCU.
Are you able to try with [ESP8266 core for Arduino](https://github.com/esp8266/Arduino)?

Alexpelillo, 2017-05-04:
But have you been able to create a new variable inside that json array that I provided? If so, just copy here the part of the code that has been able to add the variable.

Alexpelillo, 2017-05-08:
Any news? Is it related to the library?

bblanchon, 2017-05-08:
No progress, I'm sorry.
The stack traces don't make sense to me.
Can you please send an [MCVE](http://stackoverflow.com/help/mcve) that I can run on my ESP8266?

Alexpelillo, 2017-05-10:
I'll try to make a smaller piece of code and I'll paste it here.

bblanchon, 2017-05-20:
@Alexpelillo Any progress?

closed 2017-05-26 by bblanchon