bblanchon/ArduinoJson #494: No match for 'operator='

slavanau, 2017-05-02:
Hi, I have next code in my class member function:
`String jsonContent = stationFS->getJsonContent(configPath);`
 `DynamicJsonBuffer jsonBuffer;`
 `JsonObject& config = jsonBuffer.parseObject(jsonContent);`
 `SSID = config["ssid"].as<String>;`
In compiling process I got next Error

`  error: no match for 'operator=' (operand types are 'String' and '<unresolved overloaded function type>')
SSID = config["ssid"].as<String>;`

Please help me understand why. I lost lot of time with this case.

Alexpelillo, 2017-05-02:
just get the charr array like normally and then convert it to String

bblanchon, 2017-05-02:
Hi @slavanau,

The parentheses are missing.
It should be:

```c++
SSID = config["ssid"].as<String>();
```

See:

* [Error: Ambiguous overload for 'operator='](https://bblanchon.github.io/ArduinoJson/faq/how-to-fix-error-ambiguous-overload-for-operator/)

slavanau, 2017-05-03:
Thank you very much for lib and answer. As always, it's my inattention.

closed 2017-05-03 by slavanau

bblanchon, 2017-05-03:
You're welcome @slavanau
Thank you for using ArduinoJson.