bblanchon/ArduinoJson #495: ‘IsBaseOf’ is not a member of ‘ArduinoJson::TypeTraits’

Siapran, 2017-05-02:
I'm attempting to compile the headers from [the latest release](https://github.com/bblanchon/ArduinoJson/releases/download/v5.9.0/ArduinoJson-v5.9.0.h), with g++ (GCC) 6.3.1 on a 64bit linux, using C++98 standard (because I'll need to port the code to another esoteric equipment which runs a 2.6 linux kernel... so no C++11).

When attempting to compile the following code:
```cpp
#include "ArduinoJson.h"

int main(int argc, char const *argv[])
{
	/* code */
	return 0;
}
```
the compilation fails with [these errors](https://gist.github.com/Siapran/cc857c8b72abb3e120f10913ecb16a1c).

My attempts at reading the source didn't get me very far.

bblanchon, 2017-05-02:
Hi Siapran,

It's a bug in the single header file.
You can clone the repository until I fix this.

Thank you very much for reporting this issue.

Siapran, 2017-05-02:
Thank you for your answer.

I am now getting other errors:
[full logs](https://gist.github.com/Siapran/fde824df8697716a0f701fd48e0fa797)

Siapran, 2017-05-04:
I am still getting the errors from my last comment after updating the repo.

bblanchon, 2017-05-04:
Indeed, I'm trying to get my hand on a GCC 6.3.1.

As, you can see [on Continuous Integration](https://travis-ci.org/bblanchon/ArduinoJson), it works great with:

* GCC 4.4.7
* GCC 4.5.3
* GCC 4.6.4
* GCC 4.7.3
* GCC 4.8.1
* GCC 4.9.4
* GCC 5.4.1
* GCC 6.2.0
* Clang 3.4.0
* Clang 3.5.0
* Clang 3.6.2
* Clang 3.7.1
* Clang 3.8.1
* Visual Studio 2010
* Visual Studio 2012
* Visual Studio 2013
* Visual Studio 2015

Can you switch to any of these until the bug is fixed?

Siapran, 2017-05-04:
switching to GCC 4.9 gives [the same results](https://gist.github.com/Siapran/07866a82099bb56e113ce8eac4b64359)

however, I tried building the tests from the repo, and they completed without any errors, with GCC 6.3.1

maybe I'm just doing something wrong in my simple `main.cpp` file?

bblanchon, 2017-05-04:
I assumed you were still using the empty `main()` from you first message 😄 

Can you share the code?

Siapran, 2017-05-04:
[JSONTest.tar.gz](https://github.com/bblanchon/ArduinoJson/files/975726/JSONTest.tar.gz)

Siapran, 2017-05-04:
should we try a better communication medium (IM or the like, since we both speak french) so we can sort this out faster?

bblanchon, 2017-05-04:
I managed to reproduce this error by adding `-std=c++98`.
It's fixed now 🎉 

Please close this issue if compilation succeeds.

Siapran, 2017-05-05:
Compilation succeeded. Thank you for the support!

closed 2017-05-05 by Siapran

bblanchon, 2017-05-05:
You're welcome.
Thank you again for reporting these errors.
The single header will be built as part of the next release.