bblanchon/ArduinoJson #496: Decode string value with + inside issue

dmitry-rivlin-vrp, 2017-05-02:
const char defaultConfig[] PROGMEM = R"=====(
{
    "HTTP": {
    "isEnabled": true,
    "httpPort": 80,
    "fileNotFound": "/cfg/404.html",
    "mymeTypes": {
      "html": "text/html",
      "log": "plain/text",
      "css": "text/css",
      "json": "application/json",
      "gz": "application/x-gzip",
      "js": "application/javascript",
      "ico": "image/ico",
      "woff2": "font/woff2",
      "svg": "image/svg+xml",
      "png": "image/png"
    }
  }
}
)=====";

DynamicJsonBuffer jsonBuffer;

 JsonObject& jsonCfg = jsonBuffer.parseObject(defaultConfig);

when i see value in jsonCfg["HTTP"][mimeTypes"]["svg"] i see **"image/svg xml"** instead **"image/svg+xml"**

bblanchon, 2017-05-02:
Hi @dmitry-rivlin-vrp 

I was not able to reproduce this bug.
However, I saw an typo in your code: in the input it's written "mymeTypes" instead of "mimeTypes".

Can you double check ?
Regards

bblanchon, 2017-05-10:
@dmitry-rivlin-vrp any progress?

dmitry-rivlin-vrp, 2017-05-10:
I just prepared workaround.

bblanchon, 2017-05-10:
Do you still think there is a bug in the library?

dmitry-rivlin-vrp, 2017-05-11:
I am not sure. Probably i have issue with my implementation.

For example if i use \+ instead + all works as designed.

bblanchon, 2017-05-11:
I don't think the problem is related to ArduinoJson.
I'm closing this issue.
Please reopen if needed.

closed 2017-05-11 by bblanchon