bblanchon/ArduinoJson #498: Cannot call JsonVariant::as<std::string>()

Siapran, 2017-05-10:
Hi,

I'm having trouble fetching standard C++ strings from Json objects.
Getting `const char *` works fine, but attempting to call `object["field"].as<std::string>()` gives [these errors](https://gist.github.com/Siapran/f5a028c292caa1a028a706502fe6e46e) for this sample project [JSON_Config.tar.gz](https://github.com/bblanchon/ArduinoJson/files/989102/JSON_Config.tar.gz).

The relevant part takes place in `TestConfig::parse()`:
```cpp
bool TestConfig::parse( const char *configFile ) {
    CONFIG_PARSEFIELD(const char*, foo);
    CONFIG_PARSEFIELD(std::string, bar);
}
```
which looks like this after macro expansion:
```cpp
bool TestConfig::parse( const char *configFile ) {
    do { if (root->containsKey("foo") && (*root)["foo"].is<const char*>()) { foo = (*root)["foo"]; } else { throw ParseError(configFile, "Field \"" "foo" "\" is invalid."); } } while (0);
    do { if (root->containsKey("bar") && (*root)["bar"].is<std::string>()) { bar = (*root)["bar"]; } else { throw ParseError(configFile, "Field \"" "bar" "\" is invalid."); } } while (0);
}
```

The first `CONFIG_PARSEFIELD` compiles fine, unlike the second one that uses std::string.

bblanchon, 2017-05-10:
Hi Siapran,

It's not the call to `as<std::string>()` that fails; it's the call to `is<std::string>()`.
Indeed, it's [not implemented](https://bblanchon.github.io/ArduinoJson/api/jsonvariant/is/).

I'm not sure this really make sense anyway.
After all, the `JsonVariant` contains a `const char*` not a `std::string`.

Regards,
Benoit

Siapran, 2017-05-10:
Oh! I see.
Do you have any idea of how I could implement this check easily then? (keeping the `CONFIG_PARSEFIELD` interface as simple as possible)

closed 2017-05-10 by Siapran

bblanchon, 2017-05-10:
My advice it to avoid using `containsKey()`, have a look at:
https://bblanchon.github.io/ArduinoJson/api/jsonobject/containskey/