bblanchon/ArduinoJson #501: Weird compile fail when using `F()` inside lib.

sticilface, 2017-05-13:
```
Using library ArduinoJson at version 5.9.0 in folder: /Users/amelvin/Documents/Arduino/libraries/ArduinoJson 
```

the following `F()` fails to compile 
```
void SocketImpl::get(JsonArray & array)
{
		JsonObject & obj = array.createNestedObject();
		obj[F("name")] = _name;
		obj["pin"] = _pin;
		obj["state"] = _state;
}
```

with the `F()`.  If i remove the `F()` is compiles.

The error is some template kungfo

```
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:12:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
                 from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.h:3,
                 from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.cpp:1:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp: In instantiation of 'ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator ArduinoJson::JsonObject::findKey(TStringRef) [with TStringRef = const __FlashStringHelper*; ArduinoJson::Internals::List<ArduinoJson::JsonPair>::iterator = ArduinoJson::Internals::ListIterator<ArduinoJson::JsonPair>]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:314:42:   required from 'bool ArduinoJson::JsonObject::set_impl(TStringRef, TValueRef) [with TStringRef = const __FlashStringHelper*; TValueRef = const String&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:122:62:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<T>::value), bool>::type ArduinoJson::JsonObject::set(const TString*, const TValue&) [with TValue = String; TString = __FlashStringHelper; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<T>::value), bool>::type = bool]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:46:5:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), ArduinoJson::JsonObjectSubscript<TKey>&>::type ArduinoJson::JsonObjectSubscript<TKey>::operator=(const TValue&) [with TValue = String; TStringRef = const __FlashStringHelper*; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), ArduinoJson::JsonObjectSubscript<TKey>&>::type = ArduinoJson::JsonObjectSubscript<const __FlashStringHelper*>&]'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.cpp:36:18:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:295:67: error: 'equals' is not a member of 'ArduinoJson::Internals::StringTraits<const __FlashStringHelper*, void>'
       if (Internals::StringTraits<TStringRef>::equals(key, it->key)) break;
                                                                   ^
In file included from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonArray.hpp:13:0,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:11,
                 from /Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
                 from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.h:3,
                 from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.cpp:1:
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Data/ValueSetter.hpp: In instantiation of 'static bool ArduinoJson::Internals::ValueSetter<TSourceRef, Enable>::set(ArduinoJson::JsonBuffer*, TDestination&, TSourceRef) [with TDestination = const char*; TSourceRef = const __FlashStringHelper*; Enable = void]':
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:320:72:   required from 'bool ArduinoJson::JsonObject::set_impl(TStringRef, TValueRef) [with TStringRef = const __FlashStringHelper*; TValueRef = const String&]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObject.hpp:122:62:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<T>::value), bool>::type ArduinoJson::JsonObject::set(const TString*, const TValue&) [with TValue = String; TString = __FlashStringHelper; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<T>::value), bool>::type = bool]'
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonObjectSubscript.hpp:46:5:   required from 'typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), ArduinoJson::JsonObjectSubscript<TKey>&>::type ArduinoJson::JsonObjectSubscript<TKey>::operator=(const TValue&) [with TValue = String; TStringRef = const __FlashStringHelper*; typename ArduinoJson::TypeTraits::EnableIf<(! ArduinoJson::TypeTraits::IsArray<TString>::value), ArduinoJson::JsonObjectSubscript<TKey>&>::type = ArduinoJson::JsonObjectSubscript<const __FlashStringHelper*>&]'
/var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_643612/sketch/SocketImpl.cpp:36:18:   required from here
/Users/amelvin/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Data/ValueSetter.hpp:22:17: error: cannot convert 'const __FlashStringHelper*' to 'const char*' in assignment
     destination = source;
```

what is weird is if i take this function.  strip the parent class reference and place it in a sketch it compiles.  

I'm using the arduino IDE.

any ideas?

bblanchon, 2017-05-13:
That's quite strange.
What is the version of the Arduino IDE?
What is the target platform?

bblanchon, 2017-05-13:
PS: I reproduce the error when disable PROGMEM support:

```c++
#define ARDUINOJSON_ENABLE_PROGMEM 0
#include <ArduinoJson.h>

// ...
```

bblanchon, 2017-05-13:
ArduinoJson [checks if `PROGMEM` is defined](https://github.com/bblanchon/ArduinoJson/blob/master/src/ArduinoJson/Configuration.hpp#L41-L45):

```c++
// src/ArduinoJson/Configuration.hpp#L41-L45
#ifdef PROGMEM
#define ARDUINOJSON_ENABLE_PROGMEM 1
#else
#define ARDUINOJSON_ENABLE_PROGMEM 0
#endif
```

Are you sure that `PROGMEM` is supported on your platform?

sticilface, 2017-05-13:
You know what I thought about that too..  
I'm using ESP8266....  so there must be an order of things causing this... 

PROGMEM is defined

```
void SocketImpl::get(JsonArray & array)
{
	#ifndef PROGMEM
	#warning PROGMEM IS NOT DEFINED AT THIS POINT
	#endif

	JsonObject & obj = array.createNestedObject();
	obj[F("name")] = _name;
	obj["pin"] = _pin;
	obj["state"] = _state;

}
```

Adding that before does not cause the warning to be shown

sticilface, 2017-05-13:
fixed it need to include <Arduino.h> in the header of the lib, before including arduinojson.

Thanks for the fast response.

closed 2017-05-13 by sticilface