bblanchon/ArduinoJson #502: Variable name in the query

mcmega, 2017-05-13:
Hello Everybody!
Help!
I pass in the query a list of variable names. I want to get their meaning and return to JSON.

```
void GetOnRequest() {
  String root = "{}";
  DynamicJsonBuffer jb;
  JsonObject& json = jb.parseObject(root);
  for (int i = 0; i < HTTP.args(); i++) {
    json[HTTP.argName(i)] = HTTP.argName(i);
  }
  root = "";
  json.printTo(root);
  HTTP.send(200, "text/json", root);
}
```
Request type: /gor?CD=

bblanchon, 2017-05-15:
Hello @mcmega,

I'm sorry, I don't understand your question...
Please try the following pattern:

1. What did you try?
2. What is the expected outcome?
3. What result did you get instead?

mcmega, 2017-05-15:
Ok,
Request: /gor?CD=
Back I want to get the value of CD (CD - global variable in the sketch), but I'm response: {"CD":"CD"}.
I need {"CD": "Value"}
Arduino sees the CD not as the name of a variable, but as text.
The number of arguments in the query can be different, so I do not want to use "if else", but I want to cycle through the arguments. The arguments correspond to the names of the variables in the sketch.

bblanchon, 2017-05-17:
This might be one of the only acceptable use case for a [global `JsonBuffer`](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/):

```c++
// global JsonBuffer, only used for globalRepository
DynamicJsonBuffer globalJsonBuffer;

// variable repository, stores RA, TD, ND
JsonObject& globalRepository = globalJsonBuffer.createObject();

void GetOnRequest() {
  DynamicJsonBuffer jb; // <- don't reuse globalJsonBuffer
  JsonObject& json = jb.createObject();

  for (int i = 0; i < HTTP.args(); i++)
    json[HTTP.argName(i)] = globalRepositoty[HTTP.argName(i)];

  String root;
  json.printTo(root);
  HTTP.send(200, "text/json", root);
}
```

closed 2017-05-24 by bblanchon