bblanchon/ArduinoJson #503: parseObjectt fails

viktor1970, 2017-05-14:
Hallo, I've a strange issue with NodeMCU V2: I'm parsing data from WeatherUnderground; there are a lot of data, so I've a char Buf[11000] (no memory issue). The program connects to server, download data, then starts the parsing.
The issue is that only the 1st parsing is succesful, the others fails.
I create DynamicJsonBuffer inside the parsing function, and serial printing Buf shows correct data downloaded every time.
It shuld be good to have a generic GetError() function that can returns, for example, if there was memory issue or other errors.

Best regards

Vikt

bblanchon, 2017-05-15:
Hi @viktor1970,

`char Buf[11000]` is a huge red flag.
How can you be sure you don't have memory issue?

See:

* [FAQ: The first parsing succeeds, why do the next ones fail?](https://bblanchon.github.io/ArduinoJson/faq/the-first-parsing-succeeds-why-do-the-next-ones-fail/)
* [FAQ: How to reduce memory usage?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reduce-memory-usage/)
* [FAQ: Why does my device crash or reboot?](https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot/)
* [FAQ: Can I parse a JSON input that is too big to fit in memory?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

I'll add an error code in v6.

Regards,
Benoit

viktor1970, 2017-05-15:
Hallo, yes I know that the Buf is very big, but it's strange because during development it always worked for hours.
But, to avoid other issues, I deleted Buf and the parsing is done directly with http stream.
It's on test since more than 24 hours, 70 successfull parsing versus 6 failed (caused probably by connection timeout: in the night starts PC backup to the cloud). 

Best regards

Viktor

closed 2017-05-15 by viktor1970

bblanchon, 2017-05-15:
Cool 👍 
Thanks for the update.