bblanchon/ArduinoJson #504: Adding data to a File

mcmega, 2017-05-15:
Hello!
Something I'm confused ...
You need to read the json file, add the data and write back.
I did so, but I'm not sure what's right?

```
void ArbData() {
  File configFile = SPIFFS.open("/config.json","r");
  String Config = configFile.readString();
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(Config);
  for (int i=0; i < HTTP.args(); i++) {
    root[HTTP.argName(i)] = HTTP.arg(i);
  }
  configFile = SPIFFS.open("/config.json","w");
  root.printTo(configFile);
  configFile.close();
  HTTP.send(200, "text/plain", "OK");
}
```

bblanchon, 2017-05-17:
Hello Oleg,

This looks OK.
Do you have any question?

Regards,
Benoit

mcmega, 2017-05-17:
No, Thank you!

closed 2017-05-17 by bblanchon