bblanchon/ArduinoJson #506: Strange float serialization

anisimovsergey, 2017-05-18:
I looked at FAQ and Pitfalls couldn't find the answer. Please take a look at the following tests, I find it rather strange that **OneFloatDefaultDigits3** passes successfully, does it exhibit expected behavior? 

```c++
  SECTION("OneFloatDefaultDigits2") {
    array.add(0.01f);
    check(array, "[0.01]");
  }

  SECTION("OneFloatDefaultDigits3") {
    array.add(0.001f);
    check(array, "[0.00]");
  }

  SECTION("OneFloatDefaultDigits4") {
    array.add(0.0001f);
    check(array, "[10.00e-5]");
  }
```

bblanchon, 2017-05-19:
Hello Sergey,

Yes this is the expected behavior.
It's been inherited from the Arduino's `Print` class.

However, this is going to change in the next few days as I completely rewrote the float serialization.
This will be available in v5.10.0, but you can checkout branch [issue427](https://github.com/bblanchon/ArduinoJson/tree/issue427) for a preview.

Regards,
Benoit

anisimovsergey, 2017-05-19:
Hello Benoit,

Thank you for your prompt response. 

I'm not an expert in Arduiono but I find it's quite surprising that in the case of printing **0.001** I'm getting **0.00** but in case of other values like 0.01 and 0.0001 I'm getting some values (in the scientific notation or fixed point, doesn't matter much). 

I'll defensively check the preview. 

Cheers,
Sergey

anisimovsergey, 2017-05-19:
Tested it on [issue427](https://github.com/bblanchon/ArduinoJson/tree/issue427) branch - working as expected. Looking forward to seeing it in a release.

closed 2017-05-19 by anisimovsergey