bblanchon/ArduinoJson #509: Parsing json strings

7assoun21, 2017-05-21:
Hello there,
Ive read the issues asked before but I pretty much tried everything and it's not working.
Currently my sketch uses esp8266 to receive a request with a querystring that has a serialized json. The json will be an object of 3 keys with either of them filled. Sorry i couldnt attach a snipped im on mobile right now. Basically the value of a key would be an array of integers (0-255). The thing is that if that array had 36 elements in it, the parsing would work fine. However if it exceeded that, the parsing would just fail. If I then send <=36 numbers, it works normally.
Im using a DynamicJsonBuffer that it reinitialized on each request received (it was global before). I also tried static one. I tried sending strings and char arrays. Same problem, reach a certain threshold and stops parsing.
Is there a way for this?
Thank you :)

esarearthur, 2017-05-21:
I had a similar problem with static buffer define the size and add 23 to it, I have not had the issue again. Don't know why i guess the creator can help on that.

bblanchon, 2017-05-22:
Hello @7assoun21,

It looks like you're running out of memory.
Check memory usage with [ArduinoJson assistant](https://bblanchon.github.io/ArduinoJson/assistant/).

See also:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)
* [FAQ: How to reduce memory usage?](https://bblanchon.github.io/ArduinoJson/faq/how-to-reduce-memory-usage/)

Regards,
Benoit

7assoun21, 2017-05-22:
Thanks for your quick replies :)
Ill try them today when I get home and I'll send some codes.
Thanks again Ill keep you updated

7assoun21, 2017-05-22:
here's my json:
{"Animation":{"0":[0,0,0,0,0,0,0,0,0,0,0,0,0,12,0,0,0,4,16,0,0,0,0,0,0,0,0,0,0,0],"1":[0,0,0,0,0,0,0,18,65,0,0,0,0,12,0,0,0,4,16,0,0,0,0,0,0,0,0,0,0,0],"2":[0,0,0,0,0,0,0,18,65,0,0,0,0,12,0,0,0,4,16,0,0,1,66,12,0,0,0,64,0,0],"3":[0,0,0,0,34,1,0,18,65,0,0,0,32,12,0,64,0,4,17,0,0,1,66,12,0,0,0,65,0,0],"4":[0,0,8,0,34,1,0,18,65,0,0,2,32,12,0,73,0,4,81,0,0,17,66,12,0,0,16,65,32,0],"5":[0,64,8,0,34,1,0,18,65,0,0,2,32,12,0,73,0,4,81,18,0,17,66,12,0,0,16,65,36,0]},"Command":"","Mode":3,"Text":""}

I used the arudinojson assistant tool that's present there, it's a miracle! 

Thanks alot for your help guys it worked, it was a memory issue. Thanks dearest 💃

bblanchon, 2017-05-22:
You're welcome @7assoun21 , thanks for using ArduinoJson.

Don't hesitate to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2017-05-22 by bblanchon

7assoun21, 2017-05-23:
I definetly will!
Great library thanks for your efforts, you deserve it 😎