bblanchon/ArduinoJson #510: Code not compling

esarearthur, 2017-05-21:
Tried importing into atmel studio to use, but generated a lot of errors. Can this be used in atmel studio?

BR

bblanchon, 2017-05-22:
Hello Ronald,

Yes, this can be used in Atmel Studio.

Can you copy/paste the compiler output in a [gist](https://gist.github.com/)?

Regards,
Benoit

esarearthur, 2017-05-24:
[Error Generated link](https://gist.github.com/esarearthur/ae8cb63998729bc962ab61d0f70e287c)

This is blank project with only the header inserted and wont compile.

bblanchon, 2017-05-24:
These errors are in the test suite.
I don't know how Atmel Studio works, but I bet there is a way to ignore `test/` and `fuzzing/`.

esarearthur, 2017-05-24:
[Updated Error](https://gist.github.com/esarearthur/ae8cb63998729bc962ab61d0f70e287c)

I removed the test and fuzzy from the project but now it wont compile because of missing sstream used by catch

bblanchon, 2017-05-25:
Sorry Ronald, I forgot `third-party/`

However, I think you should go the "single header" way.
Instead of cloning ArduinoJson and adding the library in AtmelStudio, you just download [ArduinoJson-v5.10.0.h](https://github.com/bblanchon/ArduinoJson/releases/download/v5.10.0/ArduinoJson-v5.10.0.h) and put it in your project folder.
This is a bullet-proof technique that works with all IDEs.

bblanchon, 2017-05-30:
@esarearthur Any progress?

bblanchon, 2017-06-10:
Added [a dedicated section in the FAQ](https://bblanchon.github.io/ArduinoJson/faq/atmel-studio/)

closed 2017-06-10 by bblanchon