bblanchon/ArduinoJson #512: decimal places not working in 5.10

boblemaire, 2017-05-23:
Updated my library from 5.9.0 to 5.10.0 and set() no longer seems to use the decimal places argument.
```
stats.set("cyclerate", samplesPerCycle,0);
stats.set("chanrate",cycleSampleRate,1);
```
A response using 5.9.0:
 `{"stats":{"cyclerate":514,"chanrate":38.2,"runseconds":130,"stack":15704}}`
A response using 5.10.0: 
`{"stats":{"cyclerate":515.668701,"chanrate":39.461365,"runseconds":26,"stack":18144}}`

Both of these reference float values.  Using ESP8266/Arduino.

bblanchon, 2017-05-23:
Hello @boblemaire 

Indeed, this feature has been removed in 5.10 as it was doing more harm than good.

Was this feature important to you?
If so, please explain.

Regards,
Benoit

boblemaire, 2017-05-23:
Thanks for the quick reply. I suspected it was deliberate but it's still in the documentation so wasn't sure.  I need to limit the decimal places on these values for display purposes,  but I can do it elsewhere now that I know.

I appreciate the quality of the package and the documentation, notwithstanding this minor issue.

Regards,

Bob lemaire
Iotawatt on GitHub using ArduinoJson

bblanchon, 2017-05-23:
Sorry about that mistake, I'll update the documentation as soon as possible.
Thanks Bob.

closed 2017-05-23 by bblanchon