bblanchon/ArduinoJson #513: Can't parse

Subhajitdas298, 2017-05-24:
can't parse this.

{
"Days" : [
[[610,"S1"],[700,"S2"],[750,"S3"],[810,"S4"],[900,"S5"],[950,"S6"],[1010,"S7"]],
[[610,"S1"],[700,"S2"],[750,"S3"],[810,"S4"],[900,"S5"],[950,"S6"],[1010,"S7"]],
[[610,"S1"],[700,"S2"],[750,"S3"],[810,"S4"],[900,"S5"],[950,"S6"],[1010,"S7"]],
[[610,"S1"],[700,"S2"],[750,"S3"],[810,"S4"],[900,"S5"],[950,"S6"],[1010,"S7"]],
[[610,"S1"],[700,"S2"],[750,"S3"],[810,"S4"],[900,"S5"],[950,"S6"],[1010,"S7"]]
],
"ClassEndingTime" : 1100
}

Subhajitdas298, 2017-05-24:
Ok, got it.
Required a lot more buffrer size than actual json string.
It took me 2x buffer size.
Please inform how much size it is required in front page.

closed 2017-05-24 by Subhajitdas298

bblanchon, 2017-05-24:
Hello @Subhajitdas298,

You can use [ArduinoJson assistant](https://bblanchon.github.io/ArduinoJson/assistant/) to compute the size of the `JsonBuffer`.
The link is on the front page, in about 20 pages in the website and in 31 issues.

Please tell me exactly where you expect this information to be.

Regards,
Benoit