bblanchon/ArduinoJson #514: ESP8266 Object.printTo(FFS_File) -> empty File

Pfannex, 2017-05-26:
Hi,

I want to write back an jsonObject to my ESP8266 FFS in my own Class.
But after writing the file by using `Object.printTo(FFS_File)` the File is Empty.

On wich Point I make my mistake?

**FFSjsonFile.h**
```c++
class FFSjsonFile{
public:  
  FFSjsonFile(String filePath, int type);
  String filePath;
  size_t size;                               //filled in readJsonString()
  int type;                                  //0=Object; 1=Array
  String root;

  void loadFile();
  String readItem(String itemName);
  
private: 
  String readJsonString();
};
```

**FFSjsonFile.cpp**
```c++
FFSjsonFile::FFSjsonFile(String filePath, int type):
    filePath       (filePath),
    type           (type){
}

//  FFSjsonFile public
//===============================================================================

//===> loadFile <-----------------------------------------------------
void FFSjsonFile::loadFile(){
  root = readJsonString();
}
//===> Read json String from File <------------------------------------
String FFSjsonFile::readJsonString(){  
    
  File jsonFile;
  String jsonData = "NIL";
 
  //if (SPIFFS.begin()) {
    Serial.print("reading " + filePath + "...");
    if (SPIFFS.exists(filePath)) {
      jsonFile = SPIFFS.open(filePath, "r");
      if (jsonFile) {
        Serial.println("OK");
        size = jsonFile.size();
        std::unique_ptr<char[]> buf(new char[size]);

        jsonFile.readBytes(buf.get(), size);
        if (type == TYPE_ARRAY){
        //Array 
          DynamicJsonBuffer jsonBufferArray;
          JsonArray& jsonArray = jsonBufferArray.parseArray(buf.get());             
          if (jsonArray.success()) {
            char buffer[size];
            jsonArray.printTo(buffer, sizeof(buffer));
            jsonData = String(buffer); 
          }
        }else{
        //Object
          DynamicJsonBuffer jsonBuffer;
          JsonObject& json = jsonBuffer.parseObject(buf.get());
          if (json.success()) {         
            char buffer[size];
            json.printTo(buffer, sizeof(buffer));
            jsonData = String(buffer);
          }
        }
      }
    }
  //}
  jsonFile.close();
  return jsonData;  
}
```

**FFS.h**
```c++
class FFS{
public:
  FFS();
  FFSjsonFile myFile;
  void mount();
  
private:
};
```

**FFS.cpp**
```c++
FFS::FFS():
    //jsonFiles
    myFile(MYFILE_PATH, TYPE_OBJECT){  //global Macros

}

//===> mount FFS <-------------------------------------------------
void FFS::mount(){

  Serial.print("Mounting FS...");
  if (!SPIFFS.begin()) {
    Serial.println("Failed to mount file system");
  }else{
    Serial.println("OK");

     myFile.loadFile();
  }

//--- TEST----------------------------------------  
  Serial.println(myFile.readItem("Field_01"));
  Serial.println(myFile.readItem("Field_02"));

  //PROBLEM!
  DynamicJsonBuffer JsonBuffer;
  JsonObject& rootObject = JsonBuffer.parseObject(myFile.root);  
  rootObject["Field_01"] = "Hello";
  rootObject["Field_02"] = "World!";

  File configFile = SPIFFS.open("/MyFile.json", "w");
  if (!configFile) {
    Serial.println("Failed to open config file for writing");
  }

  rootObject.printTo(configFile);

  Serial.println(myFile.readItem("Field_01"));
  Serial.println(myFile.readItem("Field_02"));
}
```

**main.ino**
```c++
#include "FFS.h"
FFS ffs;

void setup() {  
  Serial.begin(115200); 
  Serial.println("");
  ffs.mount();  
}  

void loop() {
}
```

bblanchon, 2017-05-26:
Sorry, I have absolutely no clue.

Pfannex, 2017-05-26:
Should it work like this?
Could you verrify this problem while running the Sketch?

Pfannex, 2017-05-27:
I found my mistake......

```
        //Object
          DynamicJsonBuffer jsonBuffer;
          JsonObject& json = jsonBuffer.parseObject(buf.get());
          if (json.success()) {         
            char buffer[size+1];    //!!!!!!!
            json.printTo(buffer, sizeof(buffer));
            jsonData = String(buffer);
```
I read the the json object into a String, but the buffer was one char to smal, so the "}" was lost. That was the reason while no objet was written.

My fault, sorry....

bblanchon, 2017-05-28:
You can print directly to a `String`:

```c++
String jsonData;
json.printTo(jsonData);
```

Pfannex, 2017-05-28:
```c++
String FFSjsonFile::readJsonString(){  
    
  File jsonFile;
  String jsonData = "NIL";
 
  Serial.print("reading " + filePath + "...");
  if (SPIFFS.exists(filePath)) {
    jsonFile = SPIFFS.open(filePath, "r");
    if (jsonFile) {
      Serial.println("OK");
      size = jsonFile.size();
      std::unique_ptr<char[]> buf(new char[size]);
      jsonFile.readBytes(buf.get(), size);
      
      if (type == TYPE_ARRAY){
      //Array 
        DynamicJsonBuffer jsonBufferArray;
        JsonArray& jsonArray = jsonBufferArray.parseArray(buf.get());  
        //jsonArray.printTo(jsonData);           
        if (jsonArray.success()) {
          char buffer[size];
          jsonArray.printTo(buffer, sizeof(buffer));
          jsonData = String(buffer); 
        }
      }else{
      //Object
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        if (json.success()) {         
          //char buffer[size+1];
          //json.printTo(buffer, sizeof(buffer));
          //jsonData = String(buffer);
          json.printTo(jsonData);           
          json.printTo(Serial); 
          Serial.println("");  
          Serial.println(jsonData);        
        }
      }
    }
  }
  jsonFile.close();
  return jsonData;  
}
```
`
json.printTo(jsonData); `
this results also a empty object?

bblanchon, 2017-05-29:
ArduinoJson can read directly from a `File`:

```c++
JsonArray& jsonArray = jsonBufferArray.parseArray(jsonFile);  
```

Also, you can use `parse()` instead of `parseArray()` and `parseObject()`:

```c++
JsonVariant json = jsonBuffer.parse(jsonFile);

if (json.is<JsonArray>()) {
   JsonArray& arr = json.as<JsonArray>();
   // ...
}

if (json.is<JsonObject>()) {
   JsonObject& obj = json.as<JsonObject>();
   // ...
}
```

Pfannex, 2017-05-29:
Great hints!
I made another misstake.

`json.printTo(jsonData);`

seems to be additive, my string is initializized with "NIL".

`  String jsonData = "NIL";`

that results as string like 

> "NIL{ "webUser":"ESPuser",......."

Now it runs very well!

```c++
String FFSjsonFile::readJsonString(){  
    
  File jsonFile;
  String jsonData;
 
  Serial.print("reading " + filePath + "...");
  if (SPIFFS.exists(filePath)) {
    jsonFile = SPIFFS.open(filePath, "r");
    if (jsonFile) {
      Serial.println("OK");
      size = jsonFile.size();
      
      DynamicJsonBuffer jsonBuffer;
      JsonVariant json = jsonBuffer.parse(jsonFile);
      if (json.is<JsonArray>()) {
        JsonArray& arr = json.as<JsonArray>();
        json.printTo(jsonData);           
      }
      if (json.is<JsonObject>()) {
        JsonObject& obj = json.as<JsonObject>();
          json.printTo(jsonData);           
      }
    }else{
      Serial.println("ERROR openFile");
      return "NIL";
    }
    jsonFile.close();
  }
  return jsonData;  
}
```

The reason why I want to do this, is to read the file (access to SPIFFS) only once.
Every reuse of the file-content should be happen by using the "root-String".
Also I want to pick the Object-Items by Name to read and write them.
This features should be part of an FFS-class including differnt json-file-objects.

Do you think, that my approach is the right way to do that or am I thinking to complicated again?
Maybe this feature is implemented allready??!!

Greatings
Pf@nne

bblanchon, 2017-05-30:
Hi Pf@nne,

It seems you're trying to implement the oh-so-common global configuration object.

In my opinion, it should be implemented like that (not tested):

```c++
class Config {
  DynamicJsonBuffer _jb;
  JsonObject& _root;
public:
  Config(const char* configFile) : _jb(), _root(parse(_jb, configFile)) {

  }
  JsonVariant operator[] (const char* key) const {
     return _root[key];
  }
private:
   JsonObject& parse(JsonBuffer&, const char* configFile);
};
```

This works OK if the configuration is read-only.
However, you need to change the configuration, the `JsonBuffer` will leak.

---

I understand this class seems a good idea, since it allows to store the application configuration in a JSON file with a convenient interface.

However, I don't like this approach for the following reasons:

1. you need to store the whole JSON object tree -> waste of memory
2. you need to scan object keys each time you read a value -> waste of cpu
3. you tie all your classes with the config class -> increased coupling

In my opinion, configuration should be stored in custom `struct`s, and they should to be converted to/from JSON with dedicated functions.
As a matter of fact, this is exactly what is described in [What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library/)

I hope this helps.

Regards,
Benoit

closed 2017-05-30 by bblanchon

Pfannex, 2017-05-30:
Hi Benoit,

your hints are always helpful!
Thanks for that....

to 1.)
I need access to the whole json-tree.
So I have to decide between high SPIFFS-access and big memory variables.
In my opinion it is better to take care of the Flash.
Of course only if enough memory is available.

to 2.)
OK, a struct can save CPU-resources while direct access is available, but takes the same memory space like the json-root-string.

to 3.)
It gives me a easier access to the File-Items.

Naturaly I will consider your suggestions!

Greetings
Pf@nne