bblanchon/ArduinoJson #516: How to convert boolean (True/False) to a float value

duythang, 2017-05-31:
Dear author
I am having a problem about converting a boolean value in JSON object to a float value.
Here is my code
```
float PlugPlayMQTT::getData(const char* payload){
    StaticJsonBuffer<MQTT_MAX_PAYLOAD_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(payload);
    if (!root.success()) {
        Serial.println("parseMsg() failed");
        return;
    }
    if (root["data0"] == true){
          return 1.00;
    }else{
         return 0;
    }
}
```
I tested this code-1. I saw that when i send a true or false value, the return value is always equal to 0. 

Can you help me ? Thank you so much

duythang, 2017-05-31:
Update
I tested the code. When i send a true/false value, the return value is always 0 . When i send a float value, the return is right value.
```
float PlugPlayMQTT::getData(const char* payload, const char* type){
    StaticJsonBuffer<MQTT_MAX_PAYLOAD_SIZE> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(payload);
    if (!root.success()) {
        Serial.println("parseMsg() failed");
        return;
    }
    return root[type];
}
```

bblanchon, 2017-06-01:
Hello @duythang,

This is a bug, I may need a few days to find the time to fix it.
Meanwhile, you [can test the type of the value](https://bblanchon.github.io/ArduinoJson/faq/how-to-know-the-type-of-a-value/) before reading the content.

Sorry for the inconvenience.

Regards,
Benoit

duythang, 2017-06-02:
@bblanchon Thank you so much

closed 2017-06-03 by bblanchon