bblanchon/ArduinoJson #520: Crash on call "prettyPrintTo" or "measureLength".

dmitryordit, 2017-06-08:
The source is example to merge JSON objects from several files.
Than i uncomment line:
`Common.push_back( new jMergeFiles( "cfgtxt/enginetst.jcfg" ) );`
the Crash is occur;
The object inside enginetst.jcfg is valid. No error while parsing.
target platform: Linux Mint 4.4.0-53-generic x86_64 x86_64 x86_64
Compile in Qt (not use Qt classes or components) see attach picture.
[ArduinoJsonSend.tar.gz](https://github.com/bblanchon/ArduinoJson/files/1062502/ArduinoJsonSend.tar.gz)
[image]

bblanchon, 2017-06-09:
Hello @dmitryordit,

This is a direct consequence of the zero-copy policy: you need to ensure that the `JsonBuffer` that holds the data stays in memory.

For example, this happens when you do that:

```c++
DynamicJsonBuffer jb1;
JsonObject& obj1 = jb1.createObject();

{
    DynamicJsonBuffer jb2;
    JsonObject& obj2 = jb2.parseObject(json); 

    obj1["key"] = obj2["key"]; // now obj1 points to data in jb2
}

// now obj1["key"] contains a dangling pointer
```

This can be solved by using the same `JsonBuffer`:

```c++
DynamicJsonBuffer jb1;
JsonObject& obj1 = jb1.createObject();

JsonObject& obj2 = jb1.parseObject(json); 

obj1["key"] = obj2["key"]; // ok, points to data in jb1
```

Regards,
Benoit

dmitryordit, 2017-06-09:
After change a place the declaration of JsonBuffer outside of {}, it worked. Thanks.

bblanchon, 2017-06-10:
You're welcome 😄 
Thank you for using ArduinoJson.

If you like this probject, [please add a :star:](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2017-06-10 by bblanchon