bblanchon/ArduinoJson #522: Visual-Micro Errors

mo2menelzeiny, 2017-06-12:
I'm still experiencing the same errors as issue #483 on Visual-Micro, I tried a new empty Arduino Project with only ArduinoJson included and around 1108 errors are produced.
However by just adding #undef _MSC_VER before the ArduinoJson include statment, simply removes all the errors. I've also made sure I have the latest 5.10.0 library version.
I can't seem to understand why is this happening, any thoughts?

bblanchon, 2017-06-12:
Hello @mo2menelzeiny,

This issue is fixed on the master branch, and will be available in 5.10.1.

Regards,
Benoit

mo2menelzeiny, 2017-06-12:
Yep, just cloned it and replaced the release in the Arduino Library folder, errors are all gone! 
Thanks @bblanchon!

closed 2017-06-12 by mo2menelzeiny

bblanchon, 2017-06-12:
I just released [5.10.1](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.10.1).
Sorry about that.