bblanchon/ArduinoJson #523: Parse an element that consists of an array

gamecompiler, 2017-06-12:
I want to parse an element that consists of an array.
In this case solar->solardata->calculatedconditions->band[0-7]
How can i do this?

My attempt was:
`String data = root["solar"]["solardata"]["calculatedconditions"]["band"][0]["#text"];`
But it returns NIL

**Plattform:** ESP8266

```

{
   "solar":{
      "solardata":{
         "source":{
            "@url":"http://www.hamqsl.com/solar.html",
            "#text":"N0NBH"
         },
         "updated":"11 Jun 2017 2032 GMT",
         "solarflux":"75",
         "aindex":"4",
         "kindex":"5",
         "kindexnt":"No Report",
         "xray":"A6.8",
         "sunspots":"0",
         "heliumline":"107.7",
         "protonflux":"1.40e-01",
         "electonflux":"9.37e+01",
         "aurora":"2",
         "normalization":"1.99",
         "latdegree":"66.5",
         "solarwind":"416.2",
         "magneticfield":"-2.4",
         "calculatedconditions":{
            "band":[
               {
                  "@name":"80m-40m",
                  "@time":"day",
                  "#text":"Poor"
               },
               {
                  "@name":"30m-20m",
                  "@time":"day",
                  "#text":"Poor"
               },
               {
                  "@name":"17m-15m",
                  "@time":"day",
                  "#text":"Poor"
               },
               {
                  "@name":"12m-10m",
                  "@time":"day",
                  "#text":"Poor"
               },
               {
                  "@name":"80m-40m",
                  "@time":"night",
                  "#text":"Poor"
               },
               {
                  "@name":"30m-20m",
                  "@time":"night",
                  "#text":"Poor"
               },
               {
                  "@name":"17m-15m",
                  "@time":"night",
                  "#text":"Poor"
               },
               {
                  "@name":"12m-10m",
                  "@time":"night",
                  "#text":"Poor"
               }
            ]
         },
         "calculatedvhfconditions":{
            "phenomenon":[
               {
                  "@name":"vhf-aurora",
                  "@location":"northern_hemi",
                  "#text":"Band Closed"
               },
               {
                  "@name":"E-Skip",
                  "@location":"europe",
                  "#text":"144MHz ES"
               },
               {
                  "@name":"E-Skip",
                  "@location":"north_america",
                  "#text":"Band Closed"
               },
               {
                  "@name":"E-Skip",
                  "@location":"europe_6m",
                  "#text":"50MHz ES"
               },
               {
                  "@name":"E-Skip",
                  "@location":"europe_4m",
                  "#text":"70MHz ES"
               }
            ]
         },
         "geomagfield":"MIN STRM",
         "signalnoise":"S4-S6",
         "fof2":"NoRpt",
         "muffactor":"NoRpt",
         "muf":"NoRpt"
      }
   }
}

```

bblanchon, 2017-06-13:
Hi @gamecompiler,

`root["solar"]["solardata"]["calculatedconditions"]["band"][0]["#text"]` looks right.
Did you check that the parsing succeed?

See also:

* [FAQ: Why parsing fails?](https://bblanchon.github.io/ArduinoJson/faq/why-parsing-fails/)
* [FAQ: Parsing succeeds but I can't read the values!](https://bblanchon.github.io/ArduinoJson/faq/parsing-succeeds-but-i-cant-read-the-values/)
* [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/assistant/)

Regards,
Benoit

closed 2017-06-21 by bblanchon

gamecompiler, 2017-06-29:
Yep, the parsing failed. Php put me a blank line into the json string.
That happened because the php script tag didnt start in the first line of the script (<?php)
If ArduinoJson gets a blank line into the string, the parsing will fail.