bblanchon/ArduinoJson #528: Check for nested objects, where the hierarchy is unknown

devyte, 2017-06-17:
Target platform: ESP8266

Consider a json with nested objects, where the hierarchy is not known (i.e.: speaking in generic terms). A leaf object can then be referenced with a vector of keys.
Per this [containsKey ](https://bblanchon.github.io/ArduinoJson/api/jsonobject/containskey) link, you would do something like this for a 3-level hierarchy:

```
std::vector<String> keys = {"level1", "level2", "level3"};
int val = root[keys[0]][keys[1]][keys[2]];

```

Consider that the leaf objects aren't const chars or strings, but ints, and 0 (i.e.: zero) is a valid value. In this case, doing what the link says won't work:
```
int val = root[keys[0]][keys[1]][keys[2]];
if(val)
{
  do_something_with_val()
}
```

So what is the correct way to check for the existence of a nested object that is an arbitrary number of levels nested in, given a vector of arbitrary number of keys?

Also, how do you set the object?

I tried the following. It seems to be correct in finding the leaf objects, but then I can't set the value, so either it's luck that it's working in the search, or I'm missing something in the assignment:

```
JsonObject *
jsonContainsKeyTree(std::vector<String> & keys, JsonObject& root)
{
  JsonObject* json = &root;
  for(auto & key : keys)
  {
    if(!json->containsKey(key))  //per your recommendation, this shouldn't be used????
      return nullptr;
    JsonObject& tmp = (*json)[key];  //gets the ref
    json = &tmp; //ref to ptr
  }
  return json;
}

void somefunc()
{
...
  JsonObject *obj = jsonContainsKeyTree(keys, root); //the returned ptr seems to be correctly nullptr when a leaf is not found
  if(obj) 
  {
    //leaf object found, how do I assign a new value?
   (*obj)[key] = something; //this doesn't work
  }
...
}
```

bblanchon, 2017-06-17:
Hello @devyte,

I updated the [documentation](https://bblanchon.github.io/ArduinoJson/api/jsonobject/containskey/) to cover the case of any type of value, not just strings.
You need to use `JsonVariant` and call `.success()` to check if the variant contains a value.

Then, to find the value when you don't know the layout, I advice to recursively scan all the key-value pairs. See [Bag of Tricks: Look for a nested key](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#look-for-a-nested-key) for an example.

Regards,
Benoit

devyte, 2017-06-17:
Thanks for your answer, for some reason I hadn't seen the bag of tricks
section.

However,  the example provided there assumes that all keys in a
chaindiscussion of nestedthe objects are different, so it will return true
if any key or nested key matches, whether it's a root, node or leaf,
without regard to where in the json tree the key is found.
My use case requires matching a specific chain of nested keys to assure
that the correct leaf is found, or reported missing, because there are keys
with the same name in different parts of the tree.

Still, it showcases the use of jsonVariant, which I had to work around,
because I couldn't get it to work correctly. I think that confirms that the
solution that I found is correct.

bblanchon, 2017-06-23:
@devyte Any progress on your side?

devyte, 2017-06-23:
Yes, I pulled a trick with the parent objects to solve my problem.

It works for my usage, but I don't if it will work for generic usage. I'll
post it here later today, once I get to a PC.

devyte, 2017-06-23:
Ok, I'm finally on a normal PC again...
Consider a json like this on the ESP:
```
{
  "wificonfig":
  {
    "station":
    {
      "netinterface":
      {
        "ip":[c,c,c,c],
        "mask":[c,c,c,c],
        "gway":[c,c,c,c],
        "mac":[hh,hh,hh,hh,hh,hh]
      },
      "ssid":"somessid",
      "pass":"somepass",
      "timeout":5000
    },
    "accesspoint":
    {
      "netinterface":
      {
        "ip":[c,c,c,c],
        "mask":[c,c,c,c],
        "gway":[c,c,c,c],
        "mac":[hh,hh,hh,hh,hh,hh]
      },
      "ssid":"somessid",
      "pass","somepass",
      "maxclients":"4",
      "authtype":"WPA2",
      "channel":"3",
      "hidden","false",
      "power","82",
      "dhcpenabled":"true",
      "dhcpipstart":[c,c,c,c],
      "dhcpnumip":"50"
    },
    "opmode":"Station",
    "phymode":"n",
    "hostname","ESP-xxxxxxxx"
  }
}

```

Obviously, that holds pretty much all configurable parameters for the ESP. There are several nested levels, and there are some substructures that are repeated.
Now, consider a webserver running on the ESP, and a webclient sends a json with a partial of the above, and possibly some extra parameters that have nothing to do with the above at all.

How do I figure out which of the json objects received I am supposed to apply to the wifi configuration json?
My answer: iterate recursively over the json objects received. For each object, figure out the key chain from root to leaf. Search the ESP json config for the key chain. If it is present, then copy over the value of the received json to the ESP wifi configuration json.
If there is an error, add the error to an object parallel to the wificonfig root, and send the whole thing back.
Does that make sense? Complex, but possible to do.

Here's the catch: I have several usages that are all similar, where I have a configuration json on the ESP, and I receive a partial json from a browser, which contains some values that should be applied.
Therefore, I don't want to create one tool to do the work for each case, but rather a generic tool that will work for all cases.
That means that I don't know the hierarchies of the json objects!

devyte, 2017-06-23:
Here's what I came up with for searching for a key chain:

```
JsonObject *
jsonContainsKeyChain(std::vector<String> & keychain, JsonObject& root)
{
  JsonObject* json = &root;
  JsonObject* parent = nullptr;
  for(auto & sIt : keychain)
  {
    if(!json->containsKey(sIt))
      return nullptr;
    JsonObject& tmp = (*json)[sIt];
    parent = json;
    json = &tmp;
  }
  return parent;
}
```

A key chain is something like:
`std::vector<String> keychain = {"wificonfig", "station", "netinterface", "ip"};`
Except that I build it while iterating hierarchically through a json object.

If the key chain is found, then it the above will return the jsonObject that is the parent to the leaf. 

Why the parent? Because then I can copy the value over. Something like:
```
JsonObject * json = jsonContainsKeyChain(keychain);
(*json)[It->key] = receivedCurrentNode[It->key];
```

Why am I using json->containsKey(sIt)? Because something like this won't work:
```
int ret = (*json)[sIt];
if(!ret)
    someerror();
```

because I don't know the types, because I don't know the hierarchy, and even if I did, an int with value 0 could be a valid value, so it is indistinguishable from a failed conversion.

devyte, 2017-06-23:
Finally, how did I iterate hierarchically? This is what I came up with:

```
void
jsonUpdateConfig(
    JsonObject & currConfig, 
    JsonObject & partialConfigNode, 
    std::vector<String> & keyStack)
{
  auto oIt = partialConfigNode.begin();
  auto oEnd = partialConfigNode.end();

  if(oIt == oEnd)
  {
    //empty node, should only happen if this is called with an empty root node, and never in a recursive call
  }
  else
  {
    for (; oIt != oEnd; ++oIt)
    {
      keyStack.push_back(oIt->key);

      if(oIt->value.is<JsonObject&>())
      {
        //child is a node, follow
        jsonUpdateConfig(currConfig, oIt->value.as<JsonObject&>(), keyStack);
      }
      else
      {
        //child is a leaf, process
        JsonObject *json = jsonContainsKeyChain(keyStack, currConfig);
        if(json)
        {
          //apply config item
          (*json)[keyStack.back()] = partialConfigNode[keyStack.back()];
        }
        else
        {
          //key not found, report the error or ignore and move on

        }
      }

      keyStack.pop_back();
    }
  }
}

```

The code snippets are, of course, simplified. Also, I'm not using JsonArrays yet, so that's not taken into account.

devyte, 2017-06-23:
In the above, the currConfig object and the partialConfigNode objects are allocated in different jsonDynamicBuffers. I believe this assignment:
`(*json)[keyStack.back()] = partialConfigNode[keyStack.back()];
`
mixes objects from the two, but I'm not sure about that. I suspect I should check whether the right object is a string, and if so I should somehow duplicate it during the assignment.
@bblanchon could you please verify this?

Luckily for me, both buffers have similar lifespans right now, so it's not a problem at this time.

bblanchon, 2017-06-25:
You're absolutely right, it will mix the JsonBuffers which is usually a bad idea.
You may consider using the `clone()` function from the  [Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#clone-an-object-or-an-array)

devyte, 2017-06-29:
@bblanchon My original question is answered, so it's ok to close this :) thank you!

bblanchon, 2017-06-30:
You're welcome @devyte. Thank you for all your fast an accurate answers on other issues :+1:

closed 2017-06-30 by bblanchon