bblanchon/ArduinoJson #529: Upgrade from 5.8.4 to 5.9+ breaks makeESPArduino

aj-ptw, 2017-06-17:
Trying to upgrade above 5.10.1, but 5.9.0 introduced breaking change for makeESPArduino. 

With 5.8.4, while in the project directory of [OpenBCI Wifi](https://github.com/OpenBCI/OpenBCI_WIFI) I am able to simply type:
```bash
make -f makeEspWifiSSDP.mk
```

Steps to reproduce:
1. Get ESP266 Arduino Core from github to home dir (needed for make file)
```bash
cd ~/
git clone https://github.com/esp8266/Arduino.git esp8266
cd esp8266/tools
python get.py
```
2. Clone [OpenBCI_Wifi](www.github.com/openbci/openbci_wifi) into libraries dir
```bash
cd ~/Documents/Arduino/libraries
git clone https://github.com/OpenBCI/OpenBCI_WIFI.git
cd OpenBCI_Wifi
git checkout dev
```
3. Ensure ArduinoJSON 5.8.4 is installed to your `Documents/Arduino/libraries` dir
4. Run in terminal within `~/Documents/Arduino/libraries/OpenBCI_Wifi`
```bash
make -f makeEspWifiSSDP.mk
```
The make file should finish successfully, if it does not, comment below with error and i will answer as fast as possible with steps needed to get this compiling. Something like below is what you should be seeing:
```bash

Creating core archive
Linking /tmp/mkESP/ESP8266HuzzahSSDP_huzzah/ESP8266HuzzahSSDP.bin
  Versions: df05f81-dirty, 2.4.0-rc1-12-gb695898

Memory usage
  Ram:    56832 bytes
  Flash: 369303 bytes

Flash size: 4M (3M SPIFFS)

Build complete. Elapsed time: 5 seconds
```

Now that your make file is working as expected. Upgrade ArduinoJSON to `5.9.0`, `5.10.0`, or `5.10.1`.

4. Run in terminal within `~/Documents/Arduino/libraries/OpenBCI_Wifi`
```bash
make -f makeEspWifiSSDP.mk clean
make -f makeEspWifiSSDP.mk
```
and see error:
```
In file included from /Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.cpp:9:0:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp: In member function 'virtual bool Catch::WildcardPattern::matches(const string&) const':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:3544:52: error: exception handling disabled, use -fexceptions to enable
             throw std::logic_error( "Unknown enum" );
                                                    ^
Creating core archive
In file included from /Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.cpp:9:0:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp: At global scope:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:6414:16: error: 'stack_t' does not name a type
         static stack_t oldSigStack;
                ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:6415:33: error: 'SIGSTKSZ' was not declared in this scope
         static char altStackMem[SIGSTKSZ];
                                 ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp: In constructor 'Catch::FatalConditionHandler::FatalConditionHandler()':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:6433:13: error: 'stack_t' was not declared in this scope
             stack_t sigStack;
             ^
[... 85 lines cut ...]
                        ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7330:24: error: 'msg' was not declared in this scope
                 return msg;
                        ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7333:24: error: 'msg' was not declared in this scope
                 return msg;
                        ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7323:13: error: '...' handler must be the last handler for its try block [-fpermissive]
             catch( TestFailureException& ) {
             ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7326:13: error: '...' handler must be the last handler for its try block [-fpermissive]
             catch( std::exception& ex ) {
             ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7329:13: error: '...' handler must be the last handler for its try block [-fpermissive]
             catch( std::string& msg ) {
             ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/third-party/catch/catch.hpp:7332:13: error: '...' handler must be the last handler for its try block [-fpermissive]
             catch( const char* msg ) {
             ^
make: *** [/tmp/mkESP/ESP8266HuzzahSSDP_huzzah/catch.cpp.o] Error 1
```

I have a feeling this is from changing the name of the `/include` to `/src` but have no idea how to patch this.

This is for an ESP8266. Thanks!

aj-ptw, 2017-06-17:
Just found that if I add `/ArduinoJson/third-party` to the `EXCLUDE_DIRS` then the program compiles. Is this the solution?

aj-ptw, 2017-06-17:
I also have to have the `fuzzing` library excluded or the make file will not compile:

```
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:3:39: error: expected class-name before '{' token
 class memstream : public std::istream {

                                       ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:4:34: error: expected class-name before '{' token
   struct membuf : std::streambuf {
                                  ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp: In constructor 'memstream::membuf::membuf(const uint8_t*, size_t)':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:6:47: error: 'setg' was not declared in this scope
       setg((char *)p, (char *)p, (char *)p + l);
                                               ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp: In constructor 'memstream::memstream(const uint8_t*, size_t)':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:13:21: error: expected class-name before '(' token
       : std::istream(&_buffer), _buffer(p, l) {
                     ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:13:21: error: no matching function for call to 'memstream::membuf::membuf()'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:13:21: note: candidates are:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:5:5: note: memstream::membuf::membuf(const uint8_t*, size_t)
     membuf(const uint8_t *p, size_t l) {
     ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:5:5: note:   candidate expects 2 arguments, 0 provided
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:4:10: note: constexpr memstream::membuf::membuf(const memstream::membuf&)
   struct membuf : std::streambuf {
          ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:4:10: note:   candidate expects 1 argument, 0 provided
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:4:10: note: constexpr memstream::membuf::membuf(memstream::membuf&&)
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:4:10: note:   candidate expects 1 argument, 0 provided
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:13:21: error: expected '{' before '(' token
       : std::istream(&_buffer), _buffer(p, l) {
                     ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp: In function 'int LLVMFuzzerTestOneInput(const uint8_t*, size_t)':
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:23:16: error: 'string' is not a member of 'std'
     variant.as<std::string>();  // <- serialize to JSON
                ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:23:16: error: 'string' is not a member of 'std'
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:23:29: error: no matching function for call to 'ArduinoJson::JsonVariant::as()'
     variant.as<std::string>();  // <- serialize to JSON
                             ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:23:29: note: candidates are:
In file included from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:14:0,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/ArduinoJson.h:8,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:1:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:151:3: note: template<class T> const typename ArduinoJson::TypeTraits::EnableIf<ArduinoJson::T
ypeTraits::IsIntegral<T>::value, T>::type ArduinoJson::JsonVariant::as() const
   as() const {
   ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:151:3: note:   template argument deduction/substitution failed:
/Users/aj/Documents/Arduino/libraries/ArduinoJson/fuzzing/fuzzer.cpp:23:29: error: template argument 1 is invalid
     variant.as<std::string>();  // <- serialize to JSON
                             ^
In file included from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:14:0,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/JsonBufferBase.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.hpp:10,
                 from /Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:10,
[... cut ...]
erBuilder<TJsonBuffer, TString>::TParser ArduinoJson::Internals::makeParser(TJsonBuffer*, TString&, uint8_t)
 inline typename JsonParserBuilder<TJsonBuffer, TString>::TParser makeParser(
                                                                  ^
/Users/aj/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/JsonParser.hpp:101:66: note:   substitution of deduced template arguments resulted in errors seen above
make: *** [/tmp/mkESP/ESP8266HuzzahSSDP_huzzah/fuzzer.cpp.o] Error 1
```

bblanchon, 2017-06-17:
Hello @aj-ptw,

ArduinoJson is a header-only library, there are no `.cpp` files to compile.
All the `.cpp` files in `fuzzing/`, `test/` and `third-party/` are used for unit testing, so you should ignore them.

I don't know why your Makefile is trying to build these files.

Maybe you can try the ["single header" distribution](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.10.1) instead of cloning the repository.

Regards,
Benoit

aj-ptw, 2017-06-18:
ok thanks so much, i hope this helps someone in the future :)

closed 2017-06-18 by aj-ptw