bblanchon/ArduinoJson #532: Parsing boolean always returns false

fedor1210, 2017-06-19:
My JSON is: `{"name":"TableLED","service_name":"LED","characteristic":"On","value":true}`
And when I `bool val = root["value"];`, I get `0`.
`root["value"].is<bool>();` also returns false...

What am I doing wrong? Other objects are parsing.

_ESP8266 with latest Arduino SDK on PlatformIO._

bblanchon, 2017-06-19:
Hello @fedor1210,

Are you sure that the parsing succeeds?
Did you check `root.success()`?
Are other values (like `name` and `service_name`) OK?
Did you check with an older version of ArduinoJson?

Regards,
Benoit

fedor1210, 2017-06-19:
Good evening @bblanchon,

1. Parsing succeeds, `root.success()` returns `1`.
2. Other values are OK.
3. Tried with `5.10.1` (latest), `5.8.4` and `5.7.2`. Failed.

fedor1210, 2017-06-19:
Oh, I accidentally closed the issue :(

bblanchon, 2017-06-20:
Please see this code, it works as expected:
https://wandbox.org/permlink/Dep2QennMdQQ1cqC

Are you 100% confident with your input?

fedor1210, 2017-06-20:
Oh, I really sorry. It was the bug of ESP8266 Arduino SDK :(. Today I deleted and reinstalled (and updated) platforms on PlatformIO and it worked. Thank you for your answer!

Best wishes,
Fedor

closed 2017-06-20 by fedor1210

jscheel, 2018-01-11:
@fedor1210 can you please elaborate a bit on this? I'm experiencing the same issue.