bblanchon/ArduinoJson #533: Duplicate an entire json?

sticilface, 2017-06-20:
I'd like to copy an entire json object, duplicating all the strings and making the object entirely independent. 

example..

```cpp
char jsonStr[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
DynamicJsonBuffer buffer;
JsonObject & root = buffer.parseObject(jsonStr);

DynamicJsonBuffer buffer2;

//  some method to copy... and put root into buffer2...  

```

I've tried several methods, but not got it to work for all use cases...

bblanchon, 2017-06-21:
Hello @sticilface,

I added a `clone()` function in the [Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag-of-Tricks#clone-an-object-or-an-array).

Can we close issue #530 ?

BTW, why do you need to clone an object?

Regards,
Benoit

sticilface, 2017-06-21:
close but no cigar... this is what i had that didn't quite work.  Hadn't thought of returning the JsonVariant... 

```cpp
	if (_root.is<JsonObject&>()) { 
		JSON json(false); 
		
		for (auto obj : _root.as<JsonObject&>())
		{
			if (obj.value.is<const char *>()) {
				json.get<JsonObject&>()[obj.key] = _buffer->get().strdup(obj.value.as<const char *>()); 
			} else if ( obj.value.is<JsonObject&>() || obj.value.is<JsonArray&>() ) {
				//json.copy(obj.value); 
			} else {
				json.get<JsonObject&>()[obj.key] = obj.value; 
			}
		}
		return json; 

	} else {
		JSON json(true); 
		
		for (auto obj : _root.as<JsonArray&>())
		{
			if (obj.is<const char *>()) {
				json.get<JsonArray&>().add( _buffer->get().strdup(obj.as<const char *>())); 
			} else if ( obj.is<JsonObject&>() || obj.is<JsonArray&>() ) {
				//json.copy(obj);
			} else {
				json.get<JsonArray&>().add(obj.as<JsonVariant>()); 
			}
			
		}
```
close but no cigar.

My reason is that I've created a JSON object that manages the buffer and allows copy and move semantics, but leaving the underlying JsonBuffer the same.  So returning and passing around JSON objects does not incur any penalty.  However, this assumes that all copies and moves on my JSON object keeps the same underlying buffer, so not a true copy or move.  This is what makes it neat!  

However, there are cases where you might want to duplicate the whole thing.  Say you've passed in a stream.  This issue could be worked around but i wanted to provide the method nontheless. 

Thanks for the help.

bblanchon, 2017-06-22:
I'm closing this issue, as the conversation continued on #530

closed 2017-06-22 by bblanchon