bblanchon/ArduinoJson #534: Question - how to pass JSON object back from function?

TheOriginalMrWolf, 2017-06-21:
@bblanchon - First, thank you so much for such a fabulous library!  Using it extensively in an ESP8266 project of mine.

Unfortunately, I've run into a brick wall -- probably because I'm just learning C++ so much is still beyond me...

In summary, I'm trying to pass a JSON object (essentially a pair of KVPs containing encrypted data which, if successfully decrypted, contain another set of KVPs) to a function, have that function process the object and, under certain conditions, pass back another object (or -- less desirable -- pass the original object back with more KVPs...).  Plus I'm trying to pass objects back and forth by reference to save memory & CPU.

The actual code is quite involved, but a rough sketch of what I want to achieve is something like the following:

```
DynamicJsonBuffer encryptedRequestBuffer;
DynamicJsonBuffer plaintextRequestBuffer;
JsonObject& encryptedRoot = encryptedRequestBuffer.parseObject(encryptedData)
JsonObject& decryptedRoot;

//at this stage the encrypted root contains a JSON object that looks something like { "Ref":123, "ENCRYPTEDDATA":"<base64 encoded string containing an encrypted JSON object>"}
//The encrypted JSON object (once decrypted) contains a variety of KVPs which I don't know ahead of time, but could be, say, {"ID":12,"Value":345,"Name":"Some Numbers"}

if (decryptAndValidate( &encryptedRoot, &decryptedRoot)) {
   ...do a whole bunch of stuff with the decrypted root
}
```

decryptAndValidate:
```
decode & decrypt the encrypted root elements
Parse the decrypted data into the decryptedRoot
validate the decrypted elements
if everything ok, return to calling function with referenceToDecryptedRoot containing the decrypted JSON object.  Ideally (though not mandatory) release the memory used by the encrypted Json object
```

Um.. does that make any sense?

Tried any number of ways to do this but running into all sorts of problems like 
```
error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
```
When trying to parse the encrypted object.

Any hints or pointers in the right direction would be much appreciated!

Thanks!

TheOriginalMrWolf, 2017-06-21:
One thing contributing to my pain...

This compiles:
```
DynamicJsonBuffer _plaintextJsonRequestBuffer;
JsonObject& _decryptedRoot = _plaintextJsonRequestBuffer.parseObject("{\"Hello\":\"World\"");
```
This doesn't:
```
DynamicJsonBuffer _plaintextJsonRequestBuffer;
JsonObject& _decryptedRoot;
_decryptedRoot = _plaintextJsonRequestBuffer.parseObject("{\"Hello\":\"World\"");
```
Fails with:
```
lib\ArduinoJson/include/ArduinoJson/Data/ReferenceType.hpp:34:18: error: 'ArduinoJson::Internals::ReferenceType& ArduinoJson::Internals::ReferenceType::operator=(const ArduinoJson::Internals::ReferenceType&)' is private
```
In my utter n00bness, I thought these were semantically the same...

So, I can't defer parsing the object?

Any other way to parse an object later?

TheOriginalMrWolf, 2017-06-21:
Wellllllllll.... I fixed it.... though now my code looks like it's in pain.  Any suggestions to make this somewhat less... bizarre?

Calling function:
```
DynamicJsonBuffer _plaintextJsonRequestBuffer;
JsonObject* _decryptedRoot;
...

//root points to the root of the encrypted JSON kvp
if (cryptoEngine.aesDecryptJSONObject(response, root, _decryptedRoot, _plaintextJsonRequestBuffer)) {
   do interesting stuff
}
```

Called function:
```
bool aesDecryptJSONObject(AsyncResponseStream *response, JsonObject& root, JsonObject* &decryptedRoot, DynamicJsonBuffer&  plaintextJsonRequestBuffer);
//I LOVE JsonObject* &decryptedRoot... passing a _reference_ to a pointer to a JsonObject... OMG

...
//Do interesting stuff to decrypt the original JSON and stick it into char cryptoBuffer[]
...
//note the cast to (const char*) to force a copy of strings to the JsonRequestBuffer so that I can export out of this function
decryptedRoot = &plaintextJsonRequestBuffer.parseObject((const char*) _cryptoBufferP);
    
if ( !decryptedRoot->success()) {
//etc etc
}

//the sound you can hear is my code sobbing quietly in the corner
```

Calling function:
```
.. do stuff with: *_decryptedRoot... like...
handleSet(request, response, *_decryptedRoot);

...where the prototype for handleSet is:
void handleSet(AsyncWebServerRequest *request, AsyncResponseStream *response, JsonObject& root);

...so *_decryptedRoot gets cast back to a JsonObject& and abused accordingly...
```

If you're having trouble following, here's a picture: [fixed code working](https://img.memecdn.com/there-i-fixed-it_o_2613165.webp)

bblanchon, 2017-06-22:
Hello @TheOriginalMrWolf,

The two causes of your problems are:

1. A reference must always point to an object, and cannot be changed to point to another object
2. JsonObject cannot be copied, nor moved

On solution is to initialize the `decryptedRoot` reference and pass both objects as reference to the function.

Please try this:

```c++
bool decryptAndValidate(const JsonObject& encryptedRoot,
                        JsonObject& decryptedRoot) {
  //...
}

DynamicJsonBuffer encryptedRequestBuffer;
DynamicJsonBuffer plaintextRequestBuffer;
JsonObject& encryptedRoot = encryptedRequestBuffer.parseObject(encryptedData);
JsonObject& decryptedRoot = plaintextRequestBuffer.createObject();

if (decryptAndValidate(encryptedRoot, decryptedRoot)) {
  ... do a whole bunch of stuff with the decrypted root
}
```

Regards,
Benoit

TheOriginalMrWolf, 2017-06-23:
Hi Benoit - thank you so much for your quick response!

To be honest, this is almost exactly what I first tried.  Here is my version:
```
DynamicJsonBuffer encryptedRequestBuffer;
DynamicJsonBuffer plaintextJsonRequestBuffer;
JsonObject& root = encryptedRequestBuffer.parseObject(encryptedData);
JsonObject& decryptedRoot = plaintextJsonRequestBuffer.createObject();

if (crypto.decryptAndValidate(root, decryptedRoot, plaintextJsonRequestBuffer)) {
// etc etc
}
```
In the cryptoEngine..
```
bool cryptoEngine::decryptAndValidate(JsonObject& root, JsonObject& decryptedRoot, DynamicJsonBuffer&  plaintextJsonRequestBuffer) {
//...

 decryptedRoot = plaintextJsonRequestBuffer.parseObject((const char*) cryptoBufferP);
//...
}
```

At this point the compiler throws all the toys out of the pram with:
```
src\cryptoEngine.cpp: In member function 'bool cryptoEngine::decryptAndValidate(ArduinoJson::JsonObject&, ArduinoJson::JsonObject&, ArduinoJson::DynamicJsonBuffer&)':
src\cryptoEngine.cpp:145:19: error: use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'

...

lib\ArduinoJson/include/ArduinoJson/JsonObject.hpp:39:7: note: 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)' is implicitly deleted because the default definition would be ill-formed:
class JsonObject : public Internals::JsonPrintable<JsonObject>,

...

lib\ArduinoJson/include/ArduinoJson/Data/ReferenceType.hpp:34:18: error: 'ArduinoJson::Internals::ReferenceType& ArduinoJson::Internals::ReferenceType::operator=(const ArduinoJson::Internals::ReferenceType&)' is private
ReferenceType& operator=(const ReferenceType&);
```

Which is what started the whole saga of me poking it with pointy sticks until I could massage it into some sort of shape...

bblanchon, 2017-06-23:
Hi @TheOriginalMrWolf,

I didn't understand that you needed to call `parseObject()` within `decryptAndValidate()`.
You should consider returning the `JsonObject&`, like this:

```c++
JsonObject& cryptoEngine::decryptAndValidate(JsonObject& root, DynamicJsonBuffer&  plaintextJsonRequestBuffer) {
//...
 if (error) return JsonObject::invalid();

 JsonObject& decryptedRoot = plaintextJsonRequestBuffer.parseObject((const char*) cryptoBufferP);
//...
  return decryptedRoot;
}
```

Regards,
Benoit

TheOriginalMrWolf, 2017-06-24:
Thanks - I did that for another function and it worked fine.

Problem is that there would be quite a significant amount of refactoring required for my original function above, so (for now) am sticking to the 'reference to the pointer to' "solution".

Thing is, I don't understand _why_ passing the references to, and parsing within, my decrypt function doesn't work.  Shortcoming on my side for sure, but would really like to learn!

closed 2017-06-28 by bblanchon