bblanchon/ArduinoJson #535: Arduino UNO / Memory / Is there some kind of filter option ?

NL5L, 2017-06-21:
Hello,

I'm used to big memory's in Linux server environments, so the Arduino environment
is something new for me. My project is a 'simple' relay-board control, only the commands
will be received by MQTT in JSON.

The problem I have ("parseObject() failed") has probably to do with the 'big' JSON
messages I receive from the MQTT broker. But is there a way I can filter only two
key-value-pairs from this JSON message ? So I can save a lot of memory ?

Here is an example MQTT JSON message I am working with:

`{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014070",
   "idx" : 32,
   "name" : "Zonwering 1 (LL)",
   "nvalue" : 17,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Venetian Blinds EU",
   "unit" : 1
}`

But I am only interested in the **idx** and the **nvalue**...

This is a part of my code (I can post the whole code):

```
#include <Arduino.h>
#include <ArduinoJson.h>
#include <Ethernet.h>
#include <MqttClient.h>

[...]

void processMessage(MqttClient::MessageData& md) {
  const MqttClient::Message& msg = md.message;

  Serial.println(msg.payloadLen);
  
  char payload[msg.payloadLen];
  memcpy(payload, msg.payload, msg.payloadLen);
  payload[msg.payloadLen] = '\0';
  
  const size_t bufferSize = JSON_OBJECT_SIZE(14) + 210;
//  StaticJsonBuffer<bufferSize> jsonBuffer;   // Does not work, system seems to reboot on incomming message
  DynamicJsonBuffer jsonBuffer(bufferSize);

  Serial.println(bufferSize);
  Serial.println(payload);

  JsonObject& root = jsonBuffer.parseObject(payload);

  if (!root.success()) {
    Serial.println(F("parseObject() failed"));
//    return;
  }

  int idx = root["idx"];
  Serial.println(Domoticz_idx);
}
```

And here I have some output from the serial monitor:

```
275
354
{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014070",
   "idx" : 32,
   "name" : "Zonwering 1 (LL)",
   "nvalue" : 0,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Venetian Blinds EU",
   "unit" : 1
}

parseObject() failed
0
276
354
{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014070",
   "idx" : 32,
   "name" : "Zonwering 1 (LL)",
   "nvalue" : 17,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Venetian Blinds EU",
   "unit" : 1
}

parseObject() failed
0
275
354
{
   "Battery" : 255,
   "RSSI" : 12,
   "description" : "",
   "dtype" : "Light/Switch",
   "id" : "00014070",
   "idx" : 32,
   "name" : "Zonwering 1 (LL)",
   "nvalue" : 1,
   "stype" : "Switch",
   "svalue1" : "0",
   "switchType" : "Venetian Blinds EU",
   "unit" : 1
}

parseObject() failed
0
```

Compiler output:

```
Sketch uses 21866 bytes (67%) of program storage space. Maximum is 32256 bytes.
Global variables use 890 bytes (43%) of dynamic memory, leaving 1158 bytes for local variables. Maximum is 2048 bytes.
```

I hope I am missing something and I get my project running on a
low memory device like the UNO. Because I can not control the
JSON messages and for some future project the messages will
be even bigger. But most of the data is irrelevant... 

Kind regards

devyte, 2017-06-21:
You staticbuffer size looks tiny. Try the dynamicbuffer for a first
approach. I'm using it in a much more complicated use case, with bigger
jsons than what you show here, and it works fine.

NL5L, 2017-06-21:
Do you have some sample code ?
Because I tried many different values, and system seems to reboot on every incoming message.
Probably because the MqttClient library also uses a (lot) of memory ?!

NL5L, 2017-06-22:
When I connect to the test.mosquitto.org server I am enable to send very short JSON MQTT messages to my Arduino Uno Board. So now I can confirm it is a memory problem. The StaticJsonBuffer and the DynamicJsonBuffer both work when I set the buffer to 128.

So my original question: is there an option to filter the JSON so irrelevant data can be discard ?

bblanchon, 2017-06-22:
Hello @NL5L,

Please see:

* [FAQ: Can I parse a JSON input that is too big to fit in memory?](https://bblanchon.github.io/ArduinoJson/faq/can-i-parse-a-json-input-that-is-too-big-to-fit-in-memory/)

Regards,
Benoit

NL5L, 2017-06-22:
Thx Benoit,
I hoped there was a way to not-parse the JSON I do not need...

closed 2017-06-22 by NL5L

bblanchon, 2017-06-22:
I understand, but ArduinoJson is not the right tool for that.
Give a try to [json-streaming-parse](https://github.com/squix78/json-streaming-parser) as suggested in the FAQ.