bblanchon/ArduinoJson #536: remove () does not free the memory

hansaya, 2017-06-23:
I know that doc says dont declare a global buffer but in my case I wanted to keep certain number of records in a json object then send that out to a front end interface. So what I did was use remove(0) to remove the first item and then add a new item to the end. This helped to keep few records in history and delete the last record. However I learned that this eventually this stops working. My guess is I hit json buffer size and eventually library stop me from doing add's.

I'm sure this is not a capability of the library but why not free the memory? If I'm sending a json formatted string out every second, why should I have to initialize json buffer and use json parser to parse my data into the json format then send it out? This seems to be excessive amount of work when I can use one buffer and send out the object with new record in it.

This is my code:
```
StaticJsonBuffer<20000> jsonBuffer;
JsonObject& root = jsonBuffer.createObject();
JsonArray& timestamp = root.createNestedArray("timestamp");
int sizeHist = 10;

void addToHist() {
  unsigned long currentMillis = millis();

  if ( currentMillis - previousMillis > intervalHist )
  {
    long int tps = NTP.getTime();
    previousMillis = currentMillis;
    if ( tps > 0 )
    {
      timestamp.add(tps);
      if ( hist_t.size() > sizeHist )
      {
        timestamp.remove(0); //Delete the first record
      }
    }
  }
}
```

I'm using the latest code and running this on esp8266 with the newest release of arduinoEsp.

Is this a capability issue? Known bug with Esp? Or I'm using this library wrong? Whats your recommendation to fix this?

Thanks

devyte, 2017-06-23:
Wrong usage. The buffer is meant to hold a memory chunk. When new objects
are added, pieces of the buffer are allocated from the end. When an object
is removed, the references are removed, but the memory can't be deallocated
from the buffer. Only when the buffer itself is destroyed is the memory
returned.

Per my understanding, the use of a buffer was a design choice, hence wrong
usage.

I'm not sure exactly what you intend to do, but I suspect your mistake is
trying to keep a history of json objects. You should probably parse, and
save a history of structs/strings/vars, and not of json objects.

hansaya, 2017-06-23:
Thanks, Thats what I thought. I'm going to move my logic of saving history into an vector or list then I'll use that to make the json. Even this is done by design, might be worth freeing up some memory when a item removed. Otherwise when you call remove, you are not removing and what good does that do? Why dont we create the json without the item in the first place? if json buffer only should supposed to be short lived.

bblanchon, 2017-06-23:
@devyte's answer is accurate, as usual.
As he (or she) said, you cannot use a JsonObject to store the state of your application.

I added a big warning in the [API Reference](https://bblanchon.github.io/ArduinoJson/api/jsonobject/remove/).

See also:

* [FAQ: Why shouldn't I use a global JsonBuffer?](https://bblanchon.github.io/ArduinoJson/faq/why-shouldnt-i-use-a-global-jsonbuffer/)
* [FAQ: What's the best way to use the library?](https://bblanchon.github.io/ArduinoJson/faq/whats-the-best-way-to-use-the-library/)

hansaya, 2017-06-24:
Thanks, You guys answered my questions :)

closed 2017-06-24 by hansaya