bblanchon/ArduinoJson #538: Sample code for object access seems not to work

tius2000, 2017-06-25:
This sample code in the [API reference](https://bblanchon.github.io/ArduinoJson/api/jsonobject/containskey/) seems not to work as success() is not a member of JsonVariant:

```
JsonVariant error = root["error"];
if (error.success()) 
{
  Serial.println(error.as<char*>());
  return;
}
```

What type of JsonVariant is returned by JsonObject::get() for not exisiting keys and is there any other method to test it?

bblanchon, 2017-06-28:
Hello @tius2000,

I can assure you there is a `JsonVariant::success()`.

Have a look at:
https://wandbox.org/permlink/FPFVHsJn2cU2iNX8

You are probably using an old version of ArduinoJson.
If not, what is the compiler error?

Regards,
Benoit

tius2000, 2017-07-01:
Hi @bblanchon, thank you for your fast response and your great work! You are right, my version is out of date. Perhaps you could add JsonVariant::success() to the [this reference](https://bblanchon.github.io/ArduinoJson/api/jsonvariant/description/)?

bblanchon, 2017-07-02:
[Done](https://bblanchon.github.io/ArduinoJson/api/jsonvariant/success/).

You're welcome @tius2000, thank you for using ArduinoJson.
If you like this probject [please add a ⭐️ ](https://github.com/bblanchon/ArduinoJson/stargazers)

closed 2017-07-02 by bblanchon