bblanchon/ArduinoJson #540: Parse Json Failed and/or crash reboot

mathcoll, 2017-06-28:
Hello,

I'm trying to get a JWT token as json from a server.
The server return a correct json (format is fine).
and the Http POST response to get this json is:
> response length:3830

In my code, the buffer is like:
> const size_t bufferSize = JSON_OBJECT_SIZE(2);
But the _JsonObject& root = jsonBuffer.parseObject(lineChars); is failing_
And so I'm not able to get the JWTtoken.

I took a look to "ArduinoJson Assistant" that tells I should add:
> const size_t bufferSize = JSON_OBJECT_SIZE(2) + 3660;

But I then got exception in the Arduino (nodeMCU esp8266):
```
Exception (28):
epc1=0x401056e1 epc2=0x00000000 epc3=0x00000000 excvaddr=0x11d801c8 depc=0x00000000

ctx: sys 
sp: 3ffffdb0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff50:  00000050 00000001 40104e62 3ffed078  
3fffff60:  40206774 ffffffff 00000000 3ffe8004  
3fffff70:  4020702b ffffffff 00000000 40206f9c  
3fffff80:  3ffec5f8 40206912 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe8f88 00000001 40204ea3  
3fffffa0:  3ffe8f88 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

I guess there is something related to memory.. so I goes to FAQ: https://bblanchon.github.io/ArduinoJson/faq/why-does-my-device-crash-or-reboot/
But I'm stuck here ...

First: Am I in the correct direction? and what would you suggest me to solve issue(s)?
Thanks a lot.

devyte, 2017-06-28:
> device-crash-or-reboot/

Try the dynamicBuffer first.

mathcoll, 2017-06-28:
dynamicBuffer is working well. thank you.

closed 2017-06-28 by mathcoll