bblanchon/ArduinoJson #541: 5.11.0 breaks Homie

kylegordon, 2017-06-28:
Homie is a wifi framework for the ESP8266, and relies on ArduinoJSON. 5.11.0 causes it to break in ways that I don't fully understand. I've included some output at the bottom. It compiles correctly with 5.10.0 though, and I can't imagine a minor version change should have that effect!

This is my simple little PlatformIO project at https://github.com/kylegordon/homie-sht30

Is this something that can be sorted out between https://github.com/marvinroger/homie-esp8266/ and ArduinoJSON?

```/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_generateNetworksJson()':
/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:69:159: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'
DynamicJsonBuffer generatedJsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(1) + JSON_ARRAY_SIZE(this->_ssidCount) + (this->_ssidCount * JSON_OBJECT_SIZE(3))); // 1 at root, 3 in childrend
^
In file included from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.hpp:10:0,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.h:10,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.hpp:7,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:1:
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: note: 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)' is implicitly deleted because the default definition would be ill-formed:
class DynamicJsonBufferBase
^
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:36:7: error: use of deleted function 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)'
In file included from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:10:0,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.h:10,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.hpp:7,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:1:
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/JsonBufferBase.hpp:24:7: note: 'ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >::JsonBufferBase(const ArduinoJson::JsonBufferBase<ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator> >&)' is implicitly deleted because the default definition would be ill-formed:
class JsonBufferBase : public JsonBuffer {
^
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/JsonBufferBase.hpp:24:7: error: use of deleted function 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)'
In file included from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/JsonParser.hpp:10:0,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/JsonBufferBase.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.h:10,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.hpp:7,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:1:
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:38:7: note: 'ArduinoJson::JsonBuffer::JsonBuffer(const ArduinoJson::JsonBuffer&)' is implicitly deleted because the default definition would be ill-formed:
class JsonBuffer : Internals::NonCopyable {
^
In file included from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:14:0,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/JsonParser.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/JsonBufferBase.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.h:10,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.hpp:7,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:1:
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/../Data/NonCopyable.hpp:20:3: error: 'ArduinoJson::Internals::NonCopyable::NonCopyable(const ArduinoJson::Internals::NonCopyable&)' is private
NonCopyable(const NonCopyable&);
^
In file included from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/JsonParser.hpp:10:0,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/JsonBufferBase.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/DynamicJsonBuffer.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.hpp:10,
from /home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson.h:10,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.hpp:7,
from /home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:1:
/home/kyleg/.platformio/lib/ArduinoJson_ID64/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:38:7: error: within this context
class JsonBuffer : Internals::NonCopyable {
^
/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_onDeviceInfoRequest()':
/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootConfig.cpp:113:216: error: use of deleted function 'ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>::DynamicJsonBufferBase(const ArduinoJson::DynamicJsonBufferBase<ArduinoJson::DefaultAllocator>&)'
DynamicJsonBuffer jsonBuffer = DynamicJsonBuffer(JSON_OBJECT_SIZE(4) + JSON_OBJECT_SIZE(2) + JSON_ARRAY_SIZE(this->_interface->registeredNodesCount) + (this->_interface->registeredNodesCount * JSON_OBJECT_SIZE(2)));
^
/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootOta.cpp: In member function 'virtual void HomieInternals::BootOta::setup()':
/home/kyleg/.platformio/lib/Homie_ID555/src/Homie/Boot/BootOta.cpp:65:240: warning: 't_httpUpdate_return ESP8266HTTPUpdate::update(const String&, uint16_t, const String&, const String&, bool, const String&, bool)' is deprecated (declared at /home/kyleg/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h:82) [-Wdeprecated-declarations]
t_httpUpdate_return result = ESPhttpUpdate.update(host, port, this->_interface->config->get().ota.path, dataToPass, this->_interface->config->get().ota.server.ssl.enabled, this->_interface->config->get().ota.server.ssl.fingerprint, false);
^
Compiling .pioenvs/esp8285/lib/Homie_ID555/Homie/Helpers.o
*** [.pioenvs/esp8285/lib/Homie_ID555/Homie/Boot/BootConfig.o] Error 1
```

devyte, 2017-06-28:
> marvinroger/homie-esp8266/ and ArduinoJSON?

I believe that this is on purpose, because the buffers are now
non-copyable. I can see from the errors that there is copying of a
dynamicBuffer in the Homie code.

I'd say that Homie will have to update their code to comply.

bblanchon, 2017-06-28:
I'll take more time to review this while at home, but right now I see the following problem:

```c++
DynamicJsonBuffer jsonBuffer = DynamicJsonBuffer(/*size expression*/);
```

should be

```c++
DynamicJsonBuffer jsonBuffer(/*size expression*/);
```

bblanchon, 2017-06-28:
I'm closing this issue as I made a PR to fix this in Homie.

Kyle, thank you very much for reporting this 👍

closed 2017-06-28 by bblanchon

kylegordon, 2017-06-28:
Awesome, thank you :-)

marvinroger, 2017-06-29:
Thanks to all of you, fixed in Homie. 🎉

tigert, 2017-10-24:
Just FYI: I bumped into this when following the instructions "install homie on platformio" because "pio lib install 555" probably pulls in the latest stable release, which does not have the fix.

olskar, 2017-11-06:
Im afraid I get this error still with 5.11.2..