bblanchon/ArduinoJson #542: Not able to parse a very simple  Json string -inside a WS call back function

gnkarn, 2017-06-29:
Hi , 
after long hours and pulling my hear out of my head , and reading all i could , im not able to parse a very simple string.
on my debugging journy , i ended with a very simple setup , where the string is hard coded , and not even that way , could make it to work properly 
so my question is what other fundamental aspect am i missing ??
is Node mcu1.0 (esp8266 e12), compiling on visual studio 2015 ,

this is the relevant code 

void  wsVideoEvent(WStype_t type, uint8_t * payload, size_t length) {
	DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(3) + 100);
	Serial.printf("00 get Text: %s\r\n", payload);

	String text = ((char *)&payload[0]);
	int pos = text.indexOf('{');      // position of { json object start
	
	const char* json1 = "{\"msgName\":\"time\",\"type\":3,\"message\":\"22:40:49 GMT+0000 (UTC)\"}"; // for testing only
	JsonObject& root = jsonBuffer.parseObject(json1);
	USE_SERIAL.printf("0 json1 : %s\n", json1);
	const char* msgName = root["msgName"]; // "time"

	int wstype = root["type"]; // 3
	const char* message = root["message"]; // "22:40:49 GMT+0000 (UTC)"

and the print outputs for debuting  are : 
0 json1 : {"msgName":"time","type":3,"message":"22:40:49 GMT+0000 (UTC)"}
1 [WSc] message: pn??????msgName
2 Received from Heroku (type): 3
3 root[type]: 3

will appreciate any comment that could help to move me forward ..
Gus

bblanchon, 2017-06-30:
Hello Gus,

This is not visible in this code sample but I beleive that you are returning the string from the function.

This is a pitfall in C/C++: `const char*` do not store the string, only the memory location; you need to use `String` to make a copy.

In this case the `msgName` points to a memory location inside `DynamicJsonBuffer`, but this memory is freed as soon as the execution leaves the function. 
This is called a [danling pointer](https://en.wikipedia.org/wiki/Dangling_pointer).

Try something like that:

```c++
String msgName = root["msgName"]; // "time"
```

I hope this is helpful.

gnkarn, 2017-06-30:
thank you for your response ,

im able to parse a message , this way 

const char* json1 = "{\"msgName\":\"time\",\"type\":3,\"message\":\"22:40:49 GMT+0000 (UTC)\"}"; // for testing only																	 //USE_SERIAL.printf("0 json2: %s\n", text.c_str());

	JsonObject& root = jsonBuffer.parseObject(json1);
	if (root.success())
	{
		//Serial.println(json1);
		const char*  msgName = root["msgName"]; // "time"
		int wstype = root["type"]; // 3
		const char* message = root["message"]; // "22:40:49 GMT+0000 (UTC)"
		USE_SERIAL.printf("a [WSc] name : %s\n", msgName);
		Serial.println(wstype);
		USE_SERIAL.printf("c [WSc] message: %s\n", message);
		Serial.print("3 root[type]: ");
		Serial.println(wstype);
	}
but , as the real message i receive  has this format 
 42["time","{\"msgName\":\"time\",\"type\":3,\"message\":\"21:27:29 GMT+0000 (UTC)\"}"]

im doing  String manipulation to take , the extra characters out , before parsing de Json

String text = ((char *)&response[0]);
	int pos = text.indexOf('{');      // position of { json object start
	String json2 = text.substring(pos, text.length() - 2);  

the the String now becomes : 
{\"msgName\":\"time\",\"type\":3,\"message\":\"21:39:56 GMT+0000 (UTC)\"}

AND, in this case , ( which is the one i need) , parsing  fails  .
why ParseObject  works with const **char* json1** , and Not with a String json2.??

gnkarn, 2017-06-30:
it has to be a Way , as this is nightmare 
The problem is simple , 1) have a text that is not JSON compliant , 2) needs to cut extra substring at  the beginning to end with the true JSON string, 3) parse the resulting String .

The function PArseObject , does not accept a **S**tring as argument , and 
the arduino String class  has no way to be converted to *char ( tried with .tocharArray() function  with no luck also .

will very much appreciate your help  on making such a conversion .

devyte, 2017-07-01:
What happens if you use String::c_str()?

String lol("lollness");
Serial.println(lol.c_str());

gnkarn, 2017-07-01:
for printing  works, but the parser is not working with this string
this is the "converted" String after C_str()
{\"msgName\":\"time\",\"type\":3,\"message\":\"11:36:51 GMT+0000 (UTC)\"}

but , JsonObject& root = jsonBuffer.parseObject(json2);

is unsuccessful .

i don't know what else to try, as the input string seams perfectly formatted .
-------------
Relevant code 
<
DynamicJsonBuffer jsonBuffer(JSON_OBJECT_SIZE(3) + 100);
	// Format of date message 42["time","{\"msgName\":\"time\",\"type\":3,\"message\":\"18:15:13 GMT+0000 (UTC)\"}"]

	//String json = String ((char *)payload[0]);
	String text = ((char *)&response[0]);

	int pos = text.indexOf('{');      // position of { json object start
	String json2 = text.substring(pos, text.length() - 2) ;     //* get from "´{" to the jason obj  end
	
	json2 = json2.c_str();
	Serial.println("json2 :");
	Serial.println(json2);

	//const char* json1 = "{\"msgName\":\"time\",\"type\":3,\"message\":\"22:40:49 GMT+0000 (UTC)\"}"; // for testing only																	 //USE_SERIAL.printf("0 json2: %s\n", text.c_str());

	JsonObject& root = jsonBuffer.parseObject(json2);
	if (root.success())
	{
		//Serial.println(json1);
		const char*  msgName = root["msgName"]; // "time"
		int wstype = root["type"]; // 3
		const char* message = root["message"]; // "22:40:49 GMT+0000 (UTC)"
		USE_SERIAL.printf("a [WSc] name : %s\n", msgName);
		Serial.println(wstype);
		USE_SERIAL.printf("c [WSc] message: %s\n", message);
		Serial.print("3 root[type]: ");
		Serial.println(wstype);
	}
>

devyte, 2017-07-01:
I don't understand the problem. Does it not compile, or does it compile and
fail to parse?

gnkarn, 2017-07-03:
It compiles perfect , it just fail to parse the string.

bblanchon, 2017-07-04:
Hi @gnkarn,

I made an example with your input here:
https://wandbox.org/permlink/k2HDBLy2Ueu2e6Up

It works fine, so you should look for differences with your program.

Regards,
Benoit

devyte, 2017-07-05:
I also confirm it works for me.

bblanchon, 2017-07-09:
@gnkarn Any progress?

gnkarn, 2017-07-09:
Yes , thanks for your followup !
I found that the tricky part is that the parser is dependent on the message , 
And so it is critical to know what message you have received before parsing it ( in my case I'm receiving the message vía websockets)

The problem with plain websockets is that all messages are treated using the same callback , then 
In my case even when the websockets payload has always the same structure 
MsgName , msgType , message 

The message part is different for each 
Then they need a different parser setup

I need one for plain text message 
And other for an array of arrays 

So up to this moment , the plain text message is working fine 
And have to correctly configure the array one .

So , as soon as I have the array parser working , I would the add a mechanism on the websocket message to include a "parser" Id in the payload , that I could decode just by looking at the first char in the payload before parsing it .

Makes sense ?
:)

bblanchon, 2017-07-11:
@gnkarn, I'm not sure I really understand all you wrote 😟 
However, if you shared a **small** piece of code (or just pseudo code), I can review it.

closed 2017-07-21 by bblanchon

bblanchon, 2017-10-03:
@suryatmodulus see my answer here: https://github.com/bblanchon/ArduinoJson/issues/203#issuecomment-333805333