bblanchon/ArduinoJson #543: Precision issue in JsonFloat serialization

pgn69, 2017-07-02:
Serializing certain float numbers the result contains unexpected decimals.

Test case:
```
  StaticJsonBuffer<128> jsonBuffer;
  JsonObject& jsonObject = jsonBuffer.createObject();
  
  float x = 999.9;
  jsonObject["x"] = x;

  jsonObject.printTo(Serial);
  Serial.println();
``` 
The output is:
```
{"x":999.900024}
```

- Target platform: Bluno Nano SKU:DFR0296, based on Atmega328
- Compiler version: 4.9.2

bblanchon, 2017-07-04:
Hello @pgn69,

That behavior is expect with single precision floats.
If you need more precision, you need to with to double precision floats.
See [ARDUINOJSON_USE_DOUBLE](https://bblanchon.github.io/ArduinoJson/api/config/use_double/)

Regards,
Benoit

pgn69, 2017-07-12:
Hello,

I repeated the test with

```
#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>

```
and

```
double x = 999.9;
jsonObject["x"] = x;
jsonObject.printTo(Serial);
Serial.println();

```
but the result is the same is the same

`{"x":999.900024}`

jmpmscorp, 2017-07-13:
I'm at the same point...

I have this char array:
    `char values[] = {"32.40","32.20","32.40","32.00","31.60","31.60"}`

and i'm trying to add it to a json array.
    `
    StaticJsonBuffer<200> jsonBuffer;
    JsonObject & root = jsonBuffer.createObject();
    JsonArray & temperatura = root.createNestedArray("temperatura");

    for (byte i = 0; i < 6; i++){
      temperatura.add(strtod(values[i], NULL));
    }
    root.printTo(Serial);
`
the result is:
`{"temperatura":[32.399998,32.200001,32.399998,32,31.599999,31.599999]}`

If I add:
`#define ARDUINOJSON_USE_DOUBLE 1
#include <ArduinoJson.h>`

nothing changes...

bblanchon, 2017-07-13:
I assume you are both using 8-bit MCUs without support for double precision.

Let me see if I can reduce the number of printed digits...

jmpmscorp, 2017-07-13:
Thank you @bblanchon . Yes, I'm using an Atmega 1280P.

jmpmscorp, 2017-07-14:
More info... and something extrange...

I have been trying to achieve expected values on array but my results are confused... I have again two identical arrays like this:

~~~
char humedad[] = { "007.1951","012.0121","013.2530","008.2103","006.7865","007.5640"};
char temperatura[]={ "033.4000","033.0000","033.2000","033.0000","032.6000","032.4000"};
~~~

This arrays are converted to float, "rounded to 2 decimals" and added to JsonArray like this:

~~~
StaticJsonBuffer<200> jsonBuffer;
JsonObject & root = jsonBuffer.createObject();

JsonArray & humJson = root.createNestedArray("humedad");

for (byte i = 0; i < 6; i++){
   humJson.add( ( (int) (strtod(humedad[i], NULL) * 100 ) ) / 100.0f  );
}

JsonArray & tempJson = root.createNestedArray("temperatura");

for (byte i = 0; i < 6; i++){
   tempJson.add( ( (int) ( strtod(tempJson[i], NULL) * 100 ) ) / 100.0f  );
}

root.prinTo(Serial);
~~~

and the result is:
~~~
{"humedad":7.19,12.01,13.25,8.21,6.78,7.56],"temperatura":33.389999,33,33.200001,33,32.59,32.389999]}
~~~

I can't understand why first array is nice "rounded" and no second one.

bblanchon, 2017-07-14:
Issue fixed in master branch.

@jmpmscorp the problem is that each operation introduces an error, that's always been the problem with floating point values.

closed 2017-07-14 by bblanchon

stanleyseow, 2017-08-02:
I'm on 5.11.1 and still faces this float issue randomly, I just need 2 decimal points.

{ t1":26.56748,"t2":24.8125,"rh":30.7988 }
{ t1":26.49805,"t2":24.75,"rh":30.7454 }

I'm on Atmega2560.