bblanchon/ArduinoJson #546: Compile error on Particle Photon Firmware 0.6.2

kasperkamperman, 2017-07-07:
I'm trying to use ArduinoJSON on the Particle Photon. I use the latest stable firmware 0.6.2. 

I've included the the .h file like `#include "ArduinoJson-v5.11.0.h"` and it's stored in the application folder (I compile offline with [po-util](https://po-util.com)). 

The .hpp gave the same errors.

Minimal example:
```
#include "Particle.h"
#include "ArduinoJson-v5.11.0.h"

void setup() {

}

void loop() {
  
}
```

Error:
```
n file included from /Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/application.cpp:2:0:
/Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/ArduinoJson-v5.11.0.h: In static member function 'static bool ArduinoJson::Internals::StringTraits<const __FlashStringHelper*, void>::equals(const __FlashStringHelper*, const char*)':
/Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/ArduinoJson-v5.11.0.h:424:32: error: 'PGM_P' was not declared in this scope
     return strcmp_P(expected, (PGM_P)str) == 0;
                                ^
/Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/ArduinoJson-v5.11.0.h: In static member function 'static char* ArduinoJson::Internals::StringTraits<const __FlashStringHelper*, void>::duplicate(const __FlashStringHelper*, Buffer*)':
/Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/ArduinoJson-v5.11.0.h:429:29: error: 'PGM_P' was not declared in this scope
     size_t size = strlen_P((PGM_P)str) + 1;
                             ^
/Users/kamperman/GitHub/lumiflow-po/settingsTutorial/firmware/ArduinoJson-v5.11.0.h:431:43: error: expected ')' before 'str'
     if (dup != NULL) memcpy_P(dup, (PGM_P)str, size);
                                           ^
make[2]: *** [../build/target/user/platform-6-m/firmware/application.o] Error 1
make[1]: *** [user] Error 2
make: *** [modules/photon/user-part] Error 2
```

bblanchon, 2017-07-07:
Hi Kasper,

Please set [ARDUINOJSON_ENABLE_PROGMEM](https://bblanchon.github.io/ArduinoJson/api/config/enable_progmem/) to `0` as it seems PROGMEM isn't supported on this platform.

Regards,
Benoit

kasperkamperman, 2017-07-07:
Thanks that worked.

closed 2017-07-07 by kasperkamperman