bblanchon/ArduinoJson #547: Cannot retrieve more than one field of JSON

geovannyAvelar, 2017-07-08:
I'm receving a JSON like this when authenticating with my Spring boot API:

```
{
   "access_token":"fb0e3c9f-4387-404a-95c9-27ef721f05f2",
   "refresh_token":"1de8cee0-951d-4bd8-8356-7a92d83cce0e",
   "token_type":"bearer",
   "expires_in":181,
   "scope":"read write"
}
```

But I've tried to retrieve the access and refresh tokens, and if parse one, the other is always empty. This is my code:

```
DynamicJsonBuffer jsonBuffer;
JsonObject& jsonResponse = jsonBuffer.parseObject(stringResponse);

String accessToken = jsonResponse["access_token"].as<String>();
String refreshToken = jsonResponse["refresh_token"].as<String>();
      
Serial.println(accessToken);
Serial.println(refreshToken);
```

I'm connecting with ArduinoHttpClient (https://github.com/arduino-libraries/ArduinoHttpClient), there's nothing wrong with this lib. I've checked API logs and the requests are ok. The string with response is ok too. 

These are the messages on Serial monitor:

```
Authentication OK
{"access_token":"911df144-d2fa-4f2e-be4f-aa0ba2d63d6c","token_type":"bearer","refresh_token":"1de8cee0-951d-4bd8-8356-7a92d83cce0e","expires_in":599,"scope":"read write"}
911df144-d2fa-4f2e-be4f-aa0ba2d63d6c
```

Refresh token is empty. When i try to parse refresh before, the access token is empty instead.

bblanchon, 2017-07-09:
Hi Geovanny,

Here is an online demo with your code sample:
https://wandbox.org/permlink/QMTcsM6iAz6Nfhg4

As far as I can tell it works as expected.

Could you be running out of memory?
You could use less memory by avoiding `String`.

Regards,
Benoit

geovannyAvelar, 2017-07-15:
@bblanchon Yes, i'm facing some problems with memory. I'm running this using an Arduino UNO, and 65~75% of total memory is occupied by global variables, this is probably the root cause of my problem. I need a refactor. Thanks.

closed 2017-07-15 by geovannyAvelar

bblanchon, 2017-07-16:
You're welcome Geovanny. 
Thank you for using ArduinoJson.