bblanchon/ArduinoJson #548: Can't compile ArduinoJson with GNU++11

Sparow199, 2017-07-10:
Hello,
I'm working with [PlatformIO](https://github.com/platformio) and i'm wan't to make some unit tests with [ArduinoFake](https://github.com/FabioBatSilva/ArduinoFake).
`
[env:native]
  | platform = native
  | build_flags = -std=gnu++11
  | lib_deps = ArduinoFake
  | # lib_deps = git@github.com:FabioBatSilva/ArduinoFake.git
`
All my libraries compile without problems ( ex Hashmap for arduino ) but ArduinoJson don't.
My project is based on ArduinoJson so i can't replace it, so i can't make any unit tests.

Unfortunatly, i hadn't save the log but, it was composed by : 
`no match for ‘operator=’ (operand types are ‘ArduinoJson::JsonVariant’ and ‘const String’)`
`error: ‘equals’ is not a member of ‘ArduinoJson::Internals::StringTraits<const String&, void>`

I'm using the last realease of ArduinoJson.

Thank you.

bblanchon, 2017-07-11:
Hello @Sparow199,

Please try to set [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://bblanchon.github.io/ArduinoJson/api/config/enable_arduino_string/) to 1.

Regards,
Benoit

bblanchon, 2017-07-21:
@Sparow199, any progress?

Sparow199, 2017-07-21:
Hello, i'm sorry for late reply, the tests was in stand by waiting for solution, i tried to compile Arduino Json now with this trick and it works fine.

Thank you!

closed 2017-07-21 by Sparow199