bblanchon/ArduinoJson #550: JsonObject defined in the struct

fernandomorse, 2017-07-10:
I need to solve a problem. I want to define in a struct where a JsonObject will be stored from my json.

Json (used in parse)

{
    "Success": true,
    Payload
        "Name": "John",
        "Country": 2,
        "Raw_data": "1",
        "Created_at": "2017-07-05 22:08:42"
    },
    "Message": ""
}

I tried to use the code

Struct ResponseHttp
{
Bool success;
        JsonObject payload;
       String message;
};

Parse code
Where buffer = Json above

StaticJsonBuffer <400> jsonBuffer;
    

Char * Json = new char [buffer.length () + 1];
Memcpy (Json, buffer.c_str (), buffer.length () + 1);

JsonObject & root = jsonBuffer.parseObject (Json);
JsonObject & payload = root.get ("payload");
    
    If (! Root.success ())
    {
        _ResponseHttp = {};
    }
    Else
    {
        
        _ResponseHttp.success = root ["success"];
        _ResponseHttp.payload = root ["payload"];
        _ResponseHttp.raw_data = root ["payload"] ["raw_data"]. AsString ();
        _ResponseHttp.message = root ["message"]. AsString ();
    }
    Free (Json);
}

Is there any solution to this problem.

devyte, 2017-07-10:
This is likely a Bad Idea©. I suspect you have a design mistake.

Rather, you should store a string in the struct, and serialize your Json
object into the string. Either an Arduino String or a std::string would
work.

If you still need to store the Json object itself, then the struct will
probably need to contain the JsonBuffer that was used to parse the
JsonObject.

fernandomorse, 2017-07-11:
My project has several json responses, with format successfully payload message, where payload is variable, my initial intention was to store only the payload object for later use. I will store the buffer and test.

closed 2017-07-11 by fernandomorse