bblanchon/ArduinoJson #551: Saving strings in json buffer.

Legion102, 2017-07-11:
Hello guys,
I am working on dynamically function calling and I thought that it would be cool if I will send to the function only a reference to JsonObject(that is a part of root object) and work with it there while buffer and the root are somewhere else. But I face a problem with pointers to the static strings. 
Here is an example of code:

```
#include <ArduinoJson.h>

void msake_smth(JsonObject& data){
	char val1[] = "some text";
	data["field1"] = val1;
}

void main(){

	DynamicJsonBuffer  jsonBuffer;

	JsonObject& root = jsonBuffer.createObject();
	JsonObject& data = root.createNestedObject("data");
	msake_smth(data);

	std::string mes;
	root.prettyPrintTo(mes);
	printf(mes.c_str());

} 
```
Compiler :  Visual C++ 2013
Output:
[image]

As you can see the value of `val1` destructs and we get trash by the end. Can you advise me what to do in this situation? Is there any way to copy string value to the json buffer?

devyte, 2017-07-11:
Classic C mistake.
The json objects store pointers where possible, to reduce mem usage. Your
string is on the stack, so the object stores a pointer to a string that
gets destroyed when the function returns.

If you'll be using static strings, declare it as const char * instead of
char[ ]. Because it's const char, the string isn't on the stack, but in
memory, so the pointer is still valid after the function returns.

You could also use String or std::string.

Legion102, 2017-07-11:
Thank you for your answer.
Can you tell me what happens if we use String or std::string? Now JsonObject saves the memory instead of pointer or std::string somehow knows about the pointer and doesn't destroy the string after leaving the function?

bblanchon, 2017-07-12:
Hello @Legion102,

Indeed the behavior of ArduinoJson is different for string objects (`String` or `std::string`) and char pointers.

String objects are implicitly copied in the `JsonBuffer`, so the following would work as you expect:

```c++
void msake_smth(JsonObject& data){
	std::string val1 = "some text";
	data["field1"] = val1;
}
```

But you can also keep a `char[]` and call `JsonBuffer::strdup()` to make a copy:

```c++
void msake_smth(DynamicJsonBuffer& jb, JsonObject& data){
	char val1[] = "some text";
	data["field1"] = jb.strdup(val1);
}
```

See also:

* [Avoiding pitfalls: Do not assume that strings are copied](https://bblanchon.github.io/ArduinoJson/doc/pitfalls/#6-do-not-assume-that-strings-are-copied)
* [JsonObject::set()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/set/)
* [JsonBuffer::strdup()](https://bblanchon.github.io/ArduinoJson/api/jsonbuffer/strdup/)

Legion102, 2017-07-12:
It's clear. Thank you for answer.

closed 2017-07-12 by Legion102