bblanchon/ArduinoJson #552: Converts JsonObject to String and access

fernandomorse, 2017-07-11:
json received

{
	"success": true,
	"payload": {
		"name": "John",
                "age": 30,
		"id": 20,
		"created_at": "2017-06-19 15:11:38",
	},
	"message": "Name Created!"
}

struct Response
{	
	bool success;        
        String message;	
        const char* payload;
};

code readJSON (Where buffer is equal to json received - above)

ReadJson(String buffer)
{
    StaticJsonBuffer<400> jsonBuffer;
        
    char *Json = new char[buffer.length()+1];
    memcpy(Json, buffer.c_str(), buffer.length() + 1);

    JsonObject& root = jsonBuffer.parseObject(Json);
    JsonObject& payload = root.get("payload");
      
        
    if (!root.success())
    {
	// Failed
        Response = {};
    }
    else
    {
        //Success
        Response.payload = root["payload"];
        Response.success = root["success"];		
        Response.message = root["message"].asString();		
    }
    free(Json);           
}

How can I access the payload object contained in the Response structure?

Example

Response MyJson;
String name;
uint8_t age;

name = MyJson["name"];
age = MyJson["age"};

Please send example or code above Corrected. thanks.

devyte, 2017-07-11:
I would do something like this:
```
struct Payload
{
  String name;
  uint8_t age;
  uint32_t id;
};

struct Response
{
  bool success;
  String message;
  Payload payload;
};

ReadJson(String &buffer)
{
  StaticJsonBuffer<400> jsonBuffer;

  JsonObject& root = jsonBuffer.parseObject(buffer);
  if (!root.success())
  {
    // Failed
    Response = {}; //is this supposed to be a zeroing out of the struct??
  }
  else
  {
    //Success
    Response.success = root["success"];		
    Response.message = root["message"].asString();

    JsonObject& payload = root["payload"]; 
    Response.payload.name = payload["name"].asString();
    Response.payload.age = payload["age]; //may need to do asInt() and cast back down to uint8_t
    Response.payload.id = payload["id"];
  }
}
```

Disclaimer:
the above is untested, it may require tweaking to make it work correctly. 
Also, buffer may contain a valid json string (i.e.: parsing it will succeed), but with a wrong format or incomplete data members. Error checking should be added to protect against that.

fernandomorse, 2017-07-12:
Code implemented. Thank you
Issue closed.

closed 2017-07-12 by fernandomorse

bblanchon, 2017-07-12:
Thanks @devyte, this is a very good answer.

However,  `asInt()` and `asString()` are deprecated and should be replaced by `as<int>()` and `as<String>()`.