bblanchon/ArduinoJson #554: Strange Behavior parsing JSON

mataide, 2017-07-12:
I want to parse this JSON:

```
{
    "codeRaw": [
      "3450",
      "1700",
      "450",
      "400",
      "450",
      "1300",
      "400",
      "450",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "400",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "400",
      "450",
      "400",
      "450",
      "400",
      "1300",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "400",
      "450",
      "400",
      "450",
      "450",
      "450",
      "400",
      "1250",
      "450",
      "1300",
      "450",
      "400",
      "450",
      "1250",
      "450",
      "400",
      "450",
      "450",
      "450",
      "400",
      "450",
      "450",
      "400",
      "400",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "450",
      "1250",
      "450",
      "400",
      "450",
      "1300",
      "400",
      "1300",
      "450",
      "1250",
      "450",
      "1300",
      "450",
      "400",
      "450",
      "400",
      "450",
      "1250",
      "450",
      "450",
      "450",
      "1250",
      "450",
      "1250",
      "450",
      "450",
      "400",
      "450",
      "450",
      "400",
      "450",
      "1250",
      "450"
    ],
    "bits": 32,
    "codeType": -1,
    "kind": "IR",
    "repeatValue": false,
    "codeValue": null
  }
```
Using ArduinoJson Assistant it gives me this buffer:
```
const size_t bufferSize = JSON_ARRAY_SIZE(99) + JSON_OBJECT_SIZE(6) + 540;
DynamicJsonBuffer jsonBuffer(bufferSize);
```
If I read the JSON without parsing to JSON, it works.

But If I try to parse it, the output is:

`{"codeValue":"","bits":32,"repeatValue":false,"uuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuuu... and forever...}`

* target platform: Arduino Mega
* ide: Arduino 1.8.2 - ArduinoJson 5.11.0

mataide, 2017-07-12:
Just made a test with Arduino IDE 1.6.13 and 1.8.0, the same happens.

bblanchon, 2017-07-13:
Hello @mataide,

At first sight, it seems that the Mega is running out of memory.
This is very likely if the program keeps one or more copy of the input (1.4KB).

Can you share the relevant part of the code?

Regards,
Benoit

xzmagic, 2017-07-20:
hello,I run the deserilize function parseObject with code below
std::string jsonOutput;
char json[] = "{\"hello\":\"world\"}";
JsonObject& object = jsonBuffer.parseObject(json);
object.printTo(jsonOutput);
print(jsonOutput);

the output is awalys {}
but use the serilize function printTo can work well!
this is my first try , I use the master branch

bblanchon, 2017-07-21:
@mataide, any progress?

mataide, 2017-07-21:
@bblanchon This is probably a memory bug, I am not 100% sure but still investigating the reason. Will take a look and put here the progress.

ingrocha, 2017-07-28:
Im having a problem some times it works some time not, i get this from a websocket:
`[{"messages_campaigns_id":"1","messages_campaigns_telephone_guides_id":"6","telefono":"6674799852","mensaje":"Esto es solo una imagen de pruebas"}]`

and i get into a `parseArray` function this is my code:

```
String Sendata,Getdata;
    
    DynamicJsonBuffer jsonBuffer;
    JsonObject& rootSend = jsonBuffer.createObject();
    rootSend["type"] = "inicio";
    rootSend.printTo(Sendata);
    Serial.println(Sendata);
    jsonBuffer.clear();

    webSocketClient.sendData(Sendata);
    delay(100);
    while (webSocketClient.getData(Getdata));
    JsonArray& rootGet = jsonBuffer.parseArray(Getdata);
    if (!rootGet.success())
    {
      Serial.println("parseArray() failed");
      return;
    }
```

bblanchon, 2017-08-07:
@ingrocha I'm closing this issue as @mataide didn't provide any update. Please open a new issue if you're still facing this issue.

closed 2017-08-07 by bblanchon